JaneliaSciComp/JRCLUST #3: Plotting PSTHs

vncntprvst, 2017-12-10:
The two functions **maketrial** and **raster** do not generate anything and only return the Help and Documentation. Better documentation to use these functions and plot PSTHs in general would be appreciated. In addition, having the option to plot PSTHs from the Main GUI or the Manual GUI would be very convenient.
Thanks!

jamesjun, 2017-12-12:
Thanks for the error report. I will add a documentation on how to use this
feature.

-James

closed 2017-12-12 by vncntprvst

jamesjun, 2017-12-14:
I will update the maketrial command. I forgot to migrate this function in
the new version. You can manually set the timing by following the
instruction below:

https://github.com/JaneliaSciComp/JRCLUST/wiki/PSTH-viewing