JaneliaSciComp/JRCLUST #4: Empty "Split" window when called from Average waveform

vncntprvst, 2018-01-24:
Hi,
We have an issue when calling the Split function from the Average waveform window: most of the times, the pop up window shows only a single empty axes (the additional User dialog Yes/No/Manual shows up correctly). This is a Linux install (Ubuntu 16), an the error is found in both JRClust 3.2.4 and 3.2.5. 
Thanks.

jamesjun, 2018-02-09:
Dear Vincent,

Could you please share your recordings that show this behavior? You can post a Dropbox or Google Drive link.

vncntprvst, 2018-02-12:
Dear James,
this bug looks like it is Linux-specific, and perhaps machine-specific. I shared the data here: https://duke.box.com/s/bzlkss9kkn612yiib6th0s3cj6vsmq3y
Turns out that the display issue is just the plots being targeted to the wrong window: I see them briefly displayed in the pop up UI window (the one that asks Split Yes/No/Manual). If I close that dialog and press "S" again for the same cluster, the plots are now properly displayed in the dedicated window. That fixes the issue permanently for (at least) that session. 
However, while playing around with these plots, Matlab crashed (you'll find the dump file in the shared folder).
Thanks

enelken, 2018-04-03:
I experienced the same problem. I'm using Fedora 25 and a GeForce 1070 board. 
I solved the problem in the following way:
The gcf in resize_figure_ changes at the last two lines (following the drawnow command at the end of the function), so that the drawing in auto_split_wav_ (lines 9872 of jrc3.m and on) is performed on the wrong figure. To fix that, I changed line 9871 and following:
spfig = resize_figure_([],[0.5 0 0.5 1]);
figure(spfig);
vhAX = zeros(4,1);...
and so on. This ensures that the program draws on the correct figure.
I also experienced crashes while working with jrclust on the same machine. This was solved by shifting from hardware to software opengl.
  Eli Nelken

vncntprvst, 2019-01-21:
Hi ,
The code has changed extensively since a year ago, but that issue is still present. Here's how it looks:
[image]
Having this bug permanently fixed would be really useful. 
Thanks

closed 2019-03-11 by aliddell