JaneliaSciComp/JRCLUST #17: Error running example code. Incorrect meta file?

noahpettit, 2018-02-09:
Hi James,

I followed the installation instructions and compiling everything seemed to work.

I was also able to download the example files but ran into an issue running the example code:
`>> jrc makeprm sample.bin sample.prb`

I have copied the error message below.
It seems like it may be a problem with the sample.meta file that I have? I'm not really familiar with SpikeGL meta file format, and I don't have info about the sample recording to input information manually. I have not replaced or modified any files. Using v 3.2.5.

I will copy the error message below. Sorry if this is a simple mistake on my part, and if there is a better place to get help on issues like this (besides github issue tracker) let me know!

Thanks in advance! 

Noah

`>> jrc makeprm sample.bin sample.prb
Running 'C:\Users\harveylab\Documents\MATLAB\JRCLUST_NP\jrc3.m'
Configuration loaded from user.cfg.
<!DOCTYPE html><html lang = "en" xmlns:fb error
"http://ogp.me/ns/fb#" xml:lang = "en" class error
"media-desktop maestro" xmlns = "http://www.w3.org/1999/xhtml"> error
<head> =  error
<meta content = "IE error
edge, chrome = 1" http-equiv error
"X-UA-Compatible" /><meta content = "noindex, nofollow, noimageindex" name error
"robots" /><meta content = "origin-when-cross-origin" name error
"referrer" /><link href = "https://cfl.dropboxstatic.com/static/css/font_paper_atlas_grotesk-vflEbKJso.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/maestro_layout-vflmwFqG5.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://fonts.googleapis.com/css?family error
Open+Sans:100,200,300,400,600,700&amp;subset = latin,latin-ext" type error
"text/css" rel = "stylesheet" /> error
<script type = "text/javascript" nonce error
"4BytYlfSNWgeuowHz9Pv"> =  error
var jaguarStreaming  = jaguarStreaming || {}; error
jaguarStreaming.preloadImg  = function (url) { error
var img  = new Image(); error
img.src  = url; error
} =  error
</script><script type = "text/javascript" nonce error
"4BytYlfSNWgeuowHz9Pv"> =  error
jaguarStreaming.preloadImg("https://cfl.dropboxstatic.com/static/images/sprites/web_sprites-vflT1pLW1.png"); =  error
</script><link href = "https://cfl.dropboxstatic.com/static/css/shmodel-vflEUwGuI.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/freshbutton-vflIywrek.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/file_preview-vfllUSAzb.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/preview_flexbox_layout-vflHovmyh.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/dropbox/2015/pages/index-vflf_Uzj-.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/dropbox/2015/layout-vflvc3veE.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<script type = "text/javascript" nonce error
"4BytYlfSNWgeuowHz9Pv"> =  error
jaguarStreaming.preloadImg("https://cfl.dropboxstatic.com/static/images/header/header_logo_shmodel-vflsZ7QsE.png"); =  error
</script><script type = "text/javascript" nonce error
"4BytYlfSNWgeuowHz9Pv"> =  error
jaguarStreaming.preloadImg("https://cfl.dropboxstatic.com/static/images/sharing/error_404-vflt0X88J.png"); =  error
</script><link href = "https://cfl.dropboxstatic.com/static/css/components/login_or_register-vflAJk0Kd.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/login_or_register-vflTH-z_u.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/components/login_form-vfl3HWBvF.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/recaptcha_challenge-vflrcf67y.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/scooter/scooter-scoped-vflhQ6850.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/components/exp_cards-vflJsYU3g.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/spectrum/index.web-vflYnFJZZ.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/components/button-vfl-_t7Pp.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/legacy_packages/components-vflC9ETTX.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/accessibility-vflMYBl-j.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/recaptcha_v2_challenge-vflLfwbvt.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<script type = "text/javascript" nonce error
"4BytYlfSNWgeuowHz9Pv"> =  error
jaguarStreaming.preloadImg("https://cfl.dropboxstatic.com/static/images/icons/ajax-loading-small-vfl3Wt7C_.gif"); =  error
</script><script type = "text/javascript" nonce error
"4BytYlfSNWgeuowHz9Pv"> =  error
jaguarStreaming.preloadImg("https://cfl.dropboxstatic.com/static/images/security_keys/insert-vflkCfC4_.png"); =  error
</script><link href = "https://cfl.dropboxstatic.com/static/css/growth/shared_link_signup_modals-vflf9-K3u.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/growth/bright_modal-vfltS1uv-.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/growth/compact_form-vflfQZ6wi.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<script type = "text/javascript" nonce error
"4BytYlfSNWgeuowHz9Pv"> =  error
jaguarStreaming.preloadImg("https://cfl.dropboxstatic.com/static/images/growth/dropbox_files-vfl7phztL.png"); =  error
</script><link href = "https://cfl.dropboxstatic.com/static/css/main-vflNP2pX2.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<link href = "https://cfl.dropboxstatic.com/static/css/sprites/web_sprites-vflv2MHAO.css" type error
"text/css" crossorigin = "anonymous" rel error
"stylesheet" /> =  error
<script nonce = "4BytYlfSNWgeuowHz9Pv"> error
window._goch_  = {}; error
window.addEventListener('click', function(event) { =  error
'use strict'; =  error
for (var elm  = event.target; elm; elm  error
elm.parentElement) { =  error
if (elm.id && =  error
window._goch_.hasOwnProperty(elm.id) && =  error
window._goch_[elm.id].call(elm, event)  =  error
Index exceeds matrix dimensions.
> In C:\Users\harveylab\Documents\MATLAB\JRCLUST_NP\jrc3.m>disperr_ (line 15131)
  In C:\Users\harveylab\Documents\MATLAB\JRCLUST_NP\jrc3.m>read_meta_file_ (line 15990)
  In C:\Users\harveylab\Documents\MATLAB\JRCLUST_NP\jrc3.m>create_prm_file_ (line 15665)
  In C:\Users\harveylab\Documents\MATLAB\JRCLUST_NP\jrc3.m>makeprm_ (line 15939)
  In C:\Users\harveylab\Documents\MATLAB\JRCLUST_NP\jrc3.m>jrc3 (line 44)
  In C:\Users\harveylab\Documents\MATLAB\JRCLUST_NP\jrc.m>jrc (line 6)
Error log updated: C:\Users\harveylab\Documents\MATLAB\JRCLUST_NP\jrc_log_error.mat
read_meta_file_:
	Struct contents reference from a non-struct array object.
GPU device reset`

wolflabupenn, 2018-02-10:
Noah/James - I have the exact same error message on a new install.

Win10 for workstations, MATLAB2017A, CUDA8, VIsual Studio 2013, latest GitHub pull.

Let us know where we went wrong here.  TIA!

noahpettit, 2018-02-13:
Hi @wolflabupenn 
I found that within my 'prb' directory I had another sample.meta and sample.bin. If you open this sample.meta file, you'll see that it is a regular meta file. If you replace your other sample.meta file in your root JRCLUST directory with this file, then the step by step tutorial should work as expected. Alternatively you can manually replace the sample.meta file with one downloaded from this github repo and I think that should work. I suspect that perhaps when we run the code to download the example files, the proper meta file is being replaced with a corrupted / incorrect version. Hope that helps!

jamesjun, 2018-02-13:
Here is the documentation for SpikeGLX meta file format on Github Wiki.

https://github.com/JaneliaSciComp/JRCLUST/wiki/SpikeGLX

sample.meta from Github repo should work correctly. I will remove the zip
file from jrclust.org website which is now obsolete.

-James

-James