JaneliaSciComp/JRCLUST #20: Error calculating rho while sorting big files (Anupam Garg)

jamesjun, 2018-02-23:
Error calculating rho while sorting big files. I’m using jrclust 3.2.5 on a machine with a Quadro P5000. Here’s the error that I get:

Calculating Rho
        ........................................................................................The following error occurred converting from gpuArray to single:
Conversion to single from gpuArray is not possible.
Error in jrc3>cluster_spacetime_ (line 2280)
    vrRho(viSpk_site_) = gather_(vrRho_);
Error in jrc3>fet2clu_ (line 2195)
S_clu = cluster_spacetime_(S0, P);
Error in jrc3>sort_ (line 875)
S_clu = fet2clu_(S0, P);
Error in jrc3 (line 132)
        detect_(P); sort_(P, 0); describe_(P.vcFile_prm);

Parameters I changed from default are:
fft_thresh = 10;
qqFactor = 4.5;
blank_thresh = 10;
maxWavCor = 0.8;

Thanks for the help!

peterfschade, 2019-01-27:
I have run into this issue as well on large files in v3.2.5. Running this through the CPU also fails when it attempts:
In cuda_rho_ 
mlKeep12_ = abs(bsxfun(@minus, single(viiSpk12_ord), single(viiSpk1_ord))) <= single(dn_max);

The resulting array can be > 130gb, which causes the failure.

peterfschade, 2019-01-31:
Thank you. My files can be as large as 750 GB so I have run into this issue for most of my recordings.

aliddell, 2019-03-01:
This should be fixed as of the most recent commit.

closed 2019-03-01 by aliddell