JaneliaSciComp/JRCLUST #22: Reference to non-existent field 'mrSiteXY'

vncntprvst, 2018-03-10:
There is an undocumented 'mrSiteXY' field in the parameters struct (P), that bugs sometimes for unclear reasons.

RyanLoh2, 2018-07-05:
I am encountering this issue... have you figured out a fix/workaround for it yet?

aliddell, 2018-08-22:
Hi. Support has been in flux for JRCLUST, and we're sorry for that. Are you still having this issue? Can you post the error message you're getting?

aliddell, 2018-08-23:
Glad to hear it!

closed 2018-08-23 by aliddell

vncntprvst, 2019-01-19:
This issue is still present when running JRCLUST (current version  3.3.0) on the sample files (https://drive.google.com/drive/folders/1-UTasZWB0TwFFFV49jSrpRPHmtve34O0). E.g., `jrc makeprm sample.bin sample.meta`, then `jrc manual sample_sample.prm` returns `Reference to non-existent field 'mrSiteXY'.`

aliddell, 2019-01-23:
@vncntprvst The second argument to `makeprm` should be a probe file, but you've passed in the .meta file.

For example, using the sample data you'd have `jrc makeprm sample.bin sample.prb`. Then when you try `jrc manual sample_sample.prm` you'd get an error complaining about how `sample.bin` hasn't been sorted yet.

vncntprvst, 2019-01-29:
It makes sense, and that's what I do for my own recordings. It's just that the sample.prb is not provided with the sample data, creating confusion about how one should test, e.g., a new install. Note that `jrc unit-test` does not work (nothing happens except changing to JRClust root directory) in the current version.

aliddell, 2019-01-29:
@vncntprvst Yeah, I understand that. The way this thing was originally built was to supply a [collection of probe files](https://github.com/JaneliaSciComp/JRCLUST/tree/master/prb), and specifying a probe file that either didn't live in your current working directory or didn't have a full path would imply that your probe file was in that folder.

I also agree that the documentation is not satisfactory, but we're working on a new release which obviates a lot of what's here and have been handling questions/issues in an ad-hoc fashion. (The new release will have better docs.)

I **also** agree that the unit testing as currently done is brittle. If you don't have sample.bin in your JRCLUST folder, it silently declares victory and quits. (If you have sample.bin/sample.meta and you're still interested in testing, try moving them to the JRCLUST folder and run `jrc unit-test` again.) Again, the new release will handle this better. The alpha is out on Friday, would love to have you test it.