JaneliaSciComp/JRCLUST #24: Issue when running KiloSort

SPalestis, 2018-04-10:
I tried running KiloSort through JRC, and I got the following error. What input argument did I miss? 

[image]

SPalestis, 2018-04-10:
I figured out the previous error, the .bin file had to have the same file path.

I re-ran and got the following error:
[image]
[image]

aliddell, 2018-08-22:
Hi. Support has been in flux for JRCLUST, and we're sorry for that. If you're still interested in importing KiloSort output, we have [a feature branch](https://github.com/JaneliaSciComp/JRCLUST/tree/kilosort) (`kilosort`) which fixes these problems. This has yet to be documented correctly, but you invoke it like so:

`jrc import-ksort /path/to/your/rez.mat sessionName`

where you supply the path to your `rez.mat` file output by KiloSort and you give it a `sessionName` (which is what you would normally name your `.prm` file.

If you're not still interested in importing KiloSort output into JRCLUST, please let me know here so I can close this issue. (Even better, if you _are_ still interested, please let me know here how it works for you.)

closed 2018-10-29 by aliddell