JaneliaSciComp/JRCLUST #25: Error when running JRC on Large Experiments

gbekheet, 2018-04-16:
Hi,

I'm using JRC to detect and sort spikes in an experiment that is ~180GB. When using detect(), JRC says there are 97 operations to be completed. It moves through it nicely, but at operation 40/97 I'm given the following error:

Output argument "fGpu" (and maybe others) not assigned during call to "jrc3>wav2spk_".

[image]

JRC spikesorting worked fine on the sample experiment and a smaller experiment of ~6GB.

Any advice?

Thanks,
GB :)

RyanLoh2, 2018-11-09:
@aliddell 

I am not the OP, but I am having this same error. I am copying my parameter file below

```
% Default parameters for jrclust (default.prm)

% #Recording file format
vcFile = '';			% Raw recording file path (leave empty if multiple files are sorted together)
csFile_merge = '719_run_01_g0*.bin';			% List of recording files to merge (leave empty if single file is used)
sort_file_merge = 1;			% Sort files by the 0: alphabetical order, 1: metafile, 2: file created time, 3: file modified time, 4: numerical order in the file names
vcFile_prm = '.\719_run_01_g0_Loh_32_Chronic_Whisper.prm';			% Current parameter file path (.prm) if left empty
template_file = '';			% Template parameter file path (.prm). Parameters in this file override the default parameter file (default.prm), but get overridden is by the parameter file.
probe_file = 'Loh_32_Chronic_Whisper.prb';			% Probe file to use (.prb). Default search location: prb\
nChans = 40;			% # channels stored
vcDataType = 'int16';			% Raw data format. Choices: {'int16', 'uint16', 'int32', 'uint32', 'single', 'double'}
sRateHz = 20833;			% Sampling rate (Hz)
uV_per_bit = 0.305176;			% Bit scaling factor (uV/bit)
fTranspose_bin = 1;			% Set to 1 if the recording file is written in the channel order at different times
fInverse_file = 0;			% Set to 1 to flip the polarity of the signal
header_offset = 0;			% File header offset (set to 0 for files containing no header info (e.g. WHISPER format)

% #Execution parameters
version = 'v3.2.5';			% JRCLUST version. Updated on Jan 8, 2018
fVerbose = 0;			% Verbose flag, set to 0 to suppress displaying extra information 
fParfor = 1;			% Use Multiple CPU cores (if parallel processing toolbox is installed)
fGpu = 1;			% Use GPU if parallel processing toolbox is installed
iGpu = 1;			% Use specific GPU ID (run 'gpuDevice()' to retrieve the list of GPUs)
MAX_BYTES_LOAD = [];			% Default memory loading block size (bytes)
MAX_LOAD_SEC = [];			% Maximum loading duration (seconds). This overrides "MAX_BYTES_LOAD".
MAX_LOG = 5;			% Maximum number of histories to track for 'manual' GUI
nLoads_gpu = 8;			% Ratio of RAM to GPU memory. If your GPU memory is 12 GB and RAM is 96 GB, set nLoads_gpu = 8. Increase this number if you get GPU memory error.
vcFile_thresh = '';			% Name of .mat file taht stores the spike detection threshold in 'vnThresh_site' variable. Created by 'preview' GUI
sec_per_load_preview = 1;			% Recording duration (sec) per continuous segment to preview
nLoads_max_preview = 30;			% Number of time segments to load for preview
offset_sec_preview = 0;			% Time offset (sec) to apply to the preview
fCacheRam = 1;			% Cache spikew aveform information in RAM. Uses more RAM but faster
fRamCache = 1;			% Cache both raw and filtered spike waveforms in RAM (greater memory demand)
fSavePlot_RD = 1;			% Save the Rho-Delta plot as .png file (prefix: '_RD.png' at the end of the .prm file name)

% #Display parameters (Affects display only, (*):Can be modified in the manual GUI)
tlim = [0, 0.2];			% Time range to display 
nTime_traces = 1;			% Number of time segments to display. Set to 1 to show one continuous time segment
nSkip_show = 1;			% Show 1 in n samples when plotting traces
fSpike_show = 1;			% show spikes in the trace view (*)
maxAmp = 250;			% Amplitude range in uV (initial setting, changeable in UI) (*)
nShow = 200;			% Maximum number of traces to show [D?# spikes to show]
corrLim = [0.9, 1];			% Correlation color plot range
nSpk_show = 30;			% show spike waveforms for manual clustering
nShow_proj = 500;			% Max. number of spikes to show in the projection view
vcFet_show = 'vpp';			% Feature to show in time plot {'vpp', 'cov', 'pca'} (*)
mrColor_proj = [0.75, 0.75, 0.75; 0, 0, 0; 1, 0, 0];			% background, selected, comparison color
vcFilter_show = '';			% Filter to apply in the "traces" view. Choices: {'', 'bandpass', 'sgdiff'}. '' uses vcFilter setting.
fText = 1;			% Show spike count per unit (*, press 'n' to toggle in manual GUI)

% #Pre-processing (run "jrc spikesort" after change)
tlim_load = [];			% Time range to load in sec (uses all time range if set to empty)
vcFilter = 'ndiff';			% {'ndiff', 'sgdiff', 'bandpass', 'fir1', 'user', 'fftdiff', 'none'}
nDiff_filt = 2;			% Differentiation filter for vcFilter='sgdiff', ignored otherwise. Set to [] to disable. 2n+1 samples used for centered differentiation
freqLim = [300, 3000];			% Frequency cut-off limit for vcFilter='bandpass', ignored otherwise. 
filtOrder = 3;			% Bandpass filter order
fEllip = 1;			% Use Elliptic filter. set to 0 for Butterworth filter. Only valid for vcFilter='bandpass'.
vnFilter_user = [];			% User-specified filter kernel. Used if vcFilter='user' is set.
fft_thresh = 0;			% Automatically remove frequency outliers (unit:MAD, 10 recommended, 0 to disable). Verify by running "jrc traces" and press "p" to view the power spectrum.
freqLimNotch = {};			% Notch filter frequency range to exclude manually. {[start1, end1], [start2, end2]}. Can be used in conjunction with fft_thresh.
vcCommonRef = 'mean';			% common reference mode {'none', 'mean', 'median', 'whiten'}
viSiteZero = [];			% Site# to ignore manually
nPad_filt = 100;			% Number of samples to overlap between multiple loading (filter edge safe).
gain_boost = 1;			% multiply the raw recording by this gain to boost uV per bit
thresh_corr_bad_site = [];			% Automatically reject bad sites based on the max. correlation with neighboring sites using raw waveforms. Set to [] to disable
[... cut ...]
fDiscard_count = 1;			% Discard cluster under minimum count. set to zero to absorb to the nearest cluster
vcCluWavMode = 'mean';			% use median to compute average cluster waveform

% Batch process parameters
fProcessReverseOrder = 0;			
fProcessOdd = 0;			
fProcessEven = 0;			
fRun = 1;			% Run the job for batch

% Trial
vcFile_vid = '';			% video file
iChan_vid = [];			% video sync channel, zero-based
flim_vid = [];			% used for video sync, video frame #
tlim_vid = [];			% used for video sync, ephys time
vcFile_bonsai = '';			% bonsai 5-column output (x,y are 2 and 3)``
```

aliddell, 2018-11-12:
@RyanLoh2 thanks for the param file. I'll look into it.

aliddell, 2018-11-13:
@RyanLoh2 FYI, I haven't been able to reproduce this on my machine. Can you post:

* `ver`
* `jrc version`
* `gpuDevice()`
* `memory` (if on Windows)
* the command you ran when you got this error?

RyanLoh2, 2018-11-13:
Hi @aliddell, 
Thanks for the assistance. Answering questions below:

1) I have JRC3 (v3.2.5 - 1/8/2018) installed.
2)
```
 CUDADevice with properties:

                      Name: 'Quadro K2200'
                     Index: 1
         ComputeCapability: '5.0'
            SupportsDouble: 1
             DriverVersion: 9.2000
            ToolkitVersion: 7.5000
        MaxThreadsPerBlock: 1024
          MaxShmemPerBlock: 49152
        MaxThreadBlockSize: [1024 1024 64]
               MaxGridSize: [2.1475e+09 65535 65535]
                 SIMDWidth: 32
               TotalMemory: 4.2950e+09
           AvailableMemory: 3.8711e+09
       MultiprocessorCount: 5
              ClockRateKHz: 1124000
               ComputeMode: Default
      GPUOverlapsTransfers: 1
    KernelExecutionTimeout: 0
          CanMapHostMemory: 1
           DeviceSupported: 1
            DeviceSelected: 1
```

3) 
```
Maximum possible array:       92020 MB (9.649e+10 bytes) *
Memory available for all arrays:       92020 MB (9.649e+10 bytes) *
Memory used by MATLAB:       10020 MB (1.051e+10 bytes)
Physical Memory (RAM):       65458 MB (6.864e+10 bytes)

*  Limited by System Memory (physical + swap file) available.
```

4) The command I am running is 'JRC detect' on the loaded PRM file for the folder. The system will run through the first ~10 files before generating that error. When I use 'JRC spikesort' it will run to ~18th file before generating the same error

5) The data folder that I am accessing with the .prm is only 1.02G, so it is definitely not a size issue it seems.

Thanks again, 
Ryan

RyanLoh2, 2018-11-14:
Sure.  When I run detect() through the GUI, this is the output:  

```
Running 'D:\Projects\Users\Ryan\MATLAB\JRCLUST\jrc3.m'
Auto-set: maxSite=3.0, nSites_ref=2.0
File 1/60: detecting spikes from .\719_run_02_g0_t0.nidq.bin
Processing 1/1 of file 1/60...
	Loading from file...took 0.1s
	Filtering spikes...	took 1.4s
	Detecting spikes from each channel.
		................................
		Detected 67 spikes from 32 sites; took 0.5s.
	Merging spikes...Starting parallel pool (parpool) using the 'local' profile ... connected to 10 workers.
	26 spiking events found; took 32.3s
	Extracting features...	took 0.9s
File 1/60 took 35.4s (14.0 MB, 0.4 MB/s, x0.2 realtime)
File 2/60: detecting spikes from .\719_run_02_g0_t1.nidq.bin
Processing 1/1 of file 2/60...
	Loading from file...took 0.0s
	Filtering spikes...	took 0.0s
	Detecting spikes from each channel.
		................................
		Detected 18 spikes from 32 sites; took 0.1s.
	Merging spikes...	7 spiking events found; took 0.1s
	Extracting features...	took 0.1s
File 2/60 took 0.4s (14.0 MB, 33.2 MB/s, x16.6 realtime)
File 3/60: detecting spikes from .\719_run_02_g0_t2.nidq.bin
Processing 1/1 of file 3/60...
	Loading from file...took 0.0s
	Filtering spikes...	took 0.0s
	Detecting spikes from each channel.
		................................
		Detected 99 spikes from 32 sites; took 0.1s.
	Merging spikes...	37 spiking events found; took 0.1s
	Extracting features...	took 0.1s
File 3/60 took 0.5s (14.0 MB, 30.6 MB/s, x15.3 realtime)
File 4/60: detecting spikes from .\719_run_02_g0_t3.nidq.bin
Processing 1/1 of file 4/60...
	Loading from file...took 0.0s
	Filtering spikes...	took 0.0s
	Detecting spikes from each channel.
		................................
		Detected 27 spikes from 32 sites; took 0.1s.
	Merging spikes...	16 spiking events found; took 0.1s
	Extracting features...	took 0.1s
File 4/60 took 0.4s (14.0 MB, 38.1 MB/s, x19.0 realtime)
File 5/60: detecting spikes from .\719_run_02_g0_t4.nidq.bin
Processing 1/1 of file 5/60...
	Loading from file...took 0.0s
	Filtering spikes...	took 0.0s
	Detecting spikes from each channel.
		................................
		Detected 18 spikes from 32 sites; took 0.1s.
	Merging spikes...	5 spiking events found; took 0.1s
	Extracting features...	took 0.0s
File 5/60 took 0.3s (14.0 MB, 40.7 MB/s, x20.3 realtime)
File 6/60: detecting spikes from .\719_run_02_g0_t5.nidq.bin
Processing 1/1 of file 6/60...
	Loading from file...took 0.0s
	Filtering spikes...	took 0.0s
	Detecting spikes from each channel.
		................................
		Detected 22 spikes from 32 sites; took 0.1s.
	Merging spikes...	7 spiking events found; took 0.1s
	Extracting features...	took 0.0s
File 6/60 took 0.3s (14.0 MB, 42.4 MB/s, x21.2 realtime)
File 7/60: detecting spikes from .\719_run_02_g0_t6.nidq.bin
Processing 1/1 of file 7/60...
	Loading from file...took 0.0s
	Filtering spikes...	took 0.0s
[... 297 lines cut ...]
	Extracting features...	took 0.1s
File 17/60 took 0.3s (14.0 MB, 40.8 MB/s, x20.4 realtime)
File 18/60: detecting spikes from .\719_run_02_g0_t17.nidq.bin
Processing 1/1 of file 18/60...
	Loading from file...took 0.0s
	Filtering spikes...	took 0.0s
	Detecting spikes from each channel.
		................................
		Detected 0 spikes from 32 sites; took 0.0s.
	Merging spikes...	0 spiking events found; took 0.0s
Output argument "fGpu" (and maybe others) not assigned during call to "jrc3>wav2spk_".

Error in jrc3>file2spk_ (line 13600)
        [tnWav_raw_, tnWav_spk_, trFet_spk_, miSite_spk{end+1}, viTime_spk{end+1}, vrAmp_spk{end+1},
        vnThresh_site{end+1}, P.fGpu] ...

Error in jrc3>detect_ (line 825)
S0 = file2spk_(P);

Error in jrc3 (line 134)
        detect_(P); describe_(P.vcFile_prm);

Error in jrc (line 6)
jrc3(varargin{:});
```

RyanLoh2, 2018-11-14:
I double checked that the specific file wasn't different, t18 is the exact same size as all the other files and has it's own meta file as the others do.

aliddell, 2018-11-14:
Thanks for this. I see that JRCLUST detected no spikes in this batch (just like in the case of OP). It fails because if it detects no spikes, it returns before `fGpu` is set, so that's a bug.

I'll patch this in the master branch right away, but if you want to keep using v3.2.5, all you have to do move the statement `fGpu = P.fGpu` on line 13251 of jrc3.m to just beneath `[tnWav_spk_raw, tnWav_spk, trFet_spk, miSite_spk] = deal([]);` on line 13114. If you're familiar with patches, here's the patch:

```patch
diff --git a/jrc3.m b/jrc3.m
index b9c3193..6efed23 100644
--- a/jrc3.m
+++ b/jrc3.m
@@ -13112,6 +13112,7 @@ if nargin<5, viSite_spk = []; end
 if nargin<6, mnWav1_pre = []; end
 if nargin<7, mnWav1_post = []; end
 [tnWav_spk_raw, tnWav_spk, trFet_spk, miSite_spk] = deal([]);
+fGpu = P.fGpu;
 nFet_use = get_set_(P, 'nFet_use', 2);
 fMerge_spk = 1; %debug purpose
 fShift_pos = 0; % shift center position based on center of mass
@@ -13248,7 +13249,6 @@ end
 if nPad_pre > 0, viTime_spk = viTime_spk - nPad_pre; end
 [viTime_spk, trFet_spk, miSite_spk, tnWav_spk] = ...
     gather_(viTime_spk, trFet_spk, miSite_spk, tnWav_spk);
-fGpu = P.fGpu;
 fprintf('\ttook %0.1fs\n', toc(t_fet));
 end %func
```

Let me know if this works for you.

RyanLoh2, 2018-11-15:
This appears to have solved the issue.  Thanks!