JaneliaSciComp/JRCLUST #34: Cluster Size and Firing Rates

udlahoti, 2018-08-27:
Hi,

Our lab is currently looking at different spike sorting programs that can effectively sort high-density extracellular recordings. While we were able to get JRCLUST running, the cluster sizes are consistently 1/10th of those sorted by Klusta. 

Is there a reason JRCLUST yields lower firing rates than other softwares and if so, are there any parameters we can change to more accurately represent the dataset? Please let me know.

Thanks,
YL

aliddell, 2018-08-27:
The first thing that comes to mind is oversplitting. Are you getting comparable numbers of clusters?

udlahoti, 2018-08-27:
No, the initial number of clusters generated is almost double what Klusta outputs.

aliddell, 2018-08-27:
So you're getting fewer events detected overall, correct? Any chance I can take a look at the data and your parameter file? I could try to sort it in Klusta too, but I've never used Klusta so I probably won't get the same results.

In the meantime there are many [parameters](https://github.com/JaneliaSciComp/JRCLUST/wiki/Parameters-struct-(P)) (sorry the wiki is somewhat incomplete and out of date but this page is pretty solid, and you can sort of get an idea from the comments in your .prm file) that control how events are detected and if/how clusters are merged. You could try adjusting these. JRC is very configurable.

udlahoti, 2018-09-07:
The data can be found here: https://www.dropbox.com/sh/w6akl9mhxv1uosw/AABPIHKpt37dh4KsTgRUFPW9a?dl=0

Are there any specific parameters you recommend troubleshooting with in regards to sorting/merging?

aliddell, 2018-09-11:
First I would play with the type of common average referencing you're doing to see what kind of differences you're getting in spike counts. For instance, with `vcCommonRef` set to `mean` I saw ~59k events detected whereas with `median` I got ~62k.
You could also lower your spike detection threshold by lowering [`qqFactor`](https://github.com/JaneliaSciComp/JRCLUST/wiki/qqFactor) (applies to per-site thresholds), or you might manually set the spike detection threshold globally (I would recommend against this though).

For feature extraction, you could play with the different [features](https://github.com/JaneliaSciComp/JRCLUST/wiki/vcFet) which are supported.

For clustering, I usually like to set `nTime_clu` to 1 (for this data set I got ~62k spikes and fewer merged into others than with the default 4, and about half the clusters). You could also try lowering `maxWavCor` to, say, 0.96 or 0.97.

pinging @jamesjun for further suggestions or corrections

closed 2018-10-29 by aliddell