JaneliaSciComp/JRCLUST #41: export_quality error

jenniferColonell, 2018-09-02:
when running jrc export-quality, get the error:
-------------------------------------------------------------------
Reference to non-existent field 'nClu'.

Error in export_quality_ (line 16)
    deal((1:S.nClu)', S.vrSnr_clu(:), S.clusterSites(:), S.nSpikesPerCluster(:), ...

Error in jrc (line 195)
        case {'export-quality', 'exportquality', 'quality'}, export_quality_(P);
---------------------------------------------------------------------------

Issue is just that the field ins S_clu (subfield of S) is actually named S_clu.nClusters, not S_clu.nClu.

aliddell, 2018-09-02:
This should be fixed now -- can you check for me?

jenniferColonell, 2018-09-03:
Looks good. Thanks, Alan!

I’ve run the importer successfully now, and at least verified that the clusters get read in and exported back out w/out any changes (yes, I know you tested that!). Very nice to be able to calculate the JRC quality metrics on the kilosort clusters for comparison.

jennifer

PR #42 (2018-09-02): missed an S_clu.nClu in export_quality_
Fixes #41
closed 2018-09-03 by aliddell