JaneliaSciComp/JRCLUST #43: get0 missing in latest version?

jenniferColonell, 2018-09-03:
when running jrc spikesort...

ends w/ error (copied from command window):

'get0' is not found in the current folder or on the MATLAB path, but exists in:
    C:\Users\labadmin\Documents\JRCLUST-kilosort\JRCLUST-kilosort-082418
    C:\Users\labadmin\Documents\JRClust

Change the MATLAB current folder or add its folder to the MATLAB path.

Error in detrend_local_ (line 7)
    S0 = get0();

Error in postCluster_ (line 17)
            S_clu.clusterCenters = detrend_local_(S_clu, P, 0);

Error in fet2clu_ (line 6)
    S_clu = postCluster_(S_clu, P);

Error in sort_ (line 13)
    S_clu = fet2clu_(S0, P);

Error in jrc (line 161)
            sort_(P, 0);

aliddell, 2018-09-03:
Junchol pointed this out to me a few days ago and I fixed it in a later commit than what you have, I bet. Since changes are going to be small rapid-fire for the next little bit, it'd be worthwhile to clone the repo and pull periodically rather than downloading tarballs of the whole thing.

closed 2018-09-03 by aliddell

jenniferColonell, 2018-09-03:
Hi Alan,

I actually did check the current repo before I posted that issue, and I checked again just now – get0.m is indeed still missing in both the master and the kilosort thread. Get0_.m is present, and I haven’t studied the code to see if they do exactly the same thing (so maybe you meant to change the call rather than add get0.m back into the repo.

That said, sounds like a good idea to clone the repo instead of downloading individual downloads (which has been my practice)….

But actually, should I keep an older version for running standard JRClust stuff? Or is now a good time to be testing that along with the import functions. Let me know what’s most useful for you.

Thanks!

jennifer

aliddell, 2018-09-03:
Yup, `get0` is missing -- under the hood it was a call to `get0_` and it was only called in one place (`detrend_local_`), so I removed it. Trouble was I neglected to replace the one call in `detrend_local_` until 2748089, on Saturday.

It's useful for **me** for you and others to be finding bugs (of course this highlights the importance of testing), but if you want reliable code you're better off sticking with the master branch for now.

jenniferColonell, 2018-09-03:
Hi Alan,

OK, made a clone. Get0.m thing now fixed, as you noted.

Still running into another issue, but didn't post it as I think this is something you're working on right now--so, of course, ignore this if it's not relevant!

When running jrc manual, I get an error because the simscore (the kilosort/phy similarity score) isn't yet included in the S_Clu structure. From the command window:

Reference to non-existent field 'simScore'.

Error in plotFigClusterCor (line 38)
            figData.hImClusterCor = imagesc(S_clu.simScore, P.corrLim); %clears title and current figure

Error in manual (line 78)
    plotFigClusterCor(S0); % hFigClusterCor

Error in jrc (line 215)
        manual(P);

I can still run just by commenting out the if clause in my copy for now -- can catch up when simScore is added to sClu.

thanks,

jennifer

aliddell, 2018-09-03:
Hmm, `simScore` should belong to `S_clu` if and only if you're importing a Kilosort session. I've got a debugging tour of Janelia going on tomorrow, are you around in the morning?

jenniferColonell, 2018-09-03:
I’ve got a meeting at 11:30, but could meet with you before (say, 10:30?) or after (somewhere between 1:30 and 3…got another meeting at 3).

Let me know and I’ll make sure to be at my desk.