JaneliaSciComp/JRCLUST #45: imported kilosort data can have incorrect dimensions of spikeWaveforms

jenniferColonell, 2018-09-04:
Not sure if the issues is stale data (from previous session?) or something about the import.
Error message from command window below.
I fixed by forcing reload of spikeWaveforms in getSpikeWaveforms.m for imported data.

-----------from command window--------------------------
Unable to perform assignment because the size of the left side is 26-by-14 and the size of the right side is
26-by-10.

Error in S_clu_wav_ (line 85)
            [tmrWav_spk_clu(:, clusterSites1, iCluster), trWav_spk_clu(:, :, iCluster)] = ...

Error in S_clu_update_wav_ (line 5)
    S_clu = S_clu_wav_(S_clu);

Error in importKiloSort (line 209)
    S_clu = S_clu_update_wav_(S_clu);

Error in jrc (line 63)
            importKiloSort(arg1, arg2);

aliddell, 2018-09-06:
A fix for this is inbound, probably by the end of tomorrow.

aliddell, 2019-02-21:
Fixed a while back.

closed 2019-02-21 by aliddell