JaneliaSciComp/JRCLUST #57: Auto split

jup36, 2018-09-20:
I think there's an issue with this new auto-split scheme. So it asks how many clusters to split into, then it really gets stuck there. It just takes too long (a few minutes) until it generates the plots in the split view. Can we expedite or skip this process? This is an additional step added recently. Do people find this useful?

aliddell, 2018-09-21:
I'm not sure who finds it useful beyond its implementers, but I added it in by the request of people who wanted to see Tim et al.'s changes. In retrospect, it maybe should not be default behavior. Hopefully at the next spike sorting meeting we can discuss what's useful and what's not useful of these changes.

I'll push a patch later today which will make this an opt-in parameter.

aliddell, 2018-09-21:
Right, give the latest version a pull and let me know how that works for you.

closed 2018-09-24 by aliddell