JaneliaSciComp/JRCLUST #62: how to set uV_per_bit

yurikadoi, 2018-11-19:
When I try to make a parameter file, it asks me to put uV_per_bit value.
The default seems to be 1, but do you have any recommendations on how to set this value?
I am doing tetrode recording with openephys.
Thanks!
[image]

aliddell, 2018-11-20:
If you know the voltage range of your ADC and the gain on your amplifier, you can compute it via:
```
10^6 * (Vmax - Vmin)/2^nBits/gain
```

where `nBits` is the number of bits in your stored data type (e.g., 16 for int16)

aliddell, 2018-11-21:
Hi @yurikadoi, did this help or was there any other information I can provide?

closed 2018-11-26 by aliddell