JaneliaSciComp/JRCLUST #63: File converter for OE .continuous to SpikeGlx?

lapishc, 2018-12-18:
Does anyone have or know where to find a way to quickly and easily convert OpenEphys .continous files to the SpikeGlx format required for JRClust?

Thanks, 

ccl

aliddell, 2018-12-20:
Not sure, but it would be useful to have. I'll ask the SpikeGLX and OpenEphys people and see if they know about one.

lapishc, 2018-12-20:
Thank you, much appreciated!

aliddell, 2019-01-04:
It doesn't look like there is a converter. The OE guys [pointed me at](https://github.com/open-ephys/analysis-tools/issues/69#issuecomment-451350810) a link to a spec for the .continuous files and it looks like it shouldn't be too hard to build one, but I'm kind of swamped for the time being. Is this something you'd be willing to help with?

lapishc, 2019-01-04:
Hi Alan,

Thanks for looking into this. In principle, yes, I am willing to help but
will be pretty backed up for the next month or so.

However, I think Josh might have undersold the task a bit. If memory serves
(often it does not!) I think there might be quite a bit of metadata that
needs to be written into the header.

At any rate, I will be back in touch when I start to monkey with it.

Thanks again for the help,

ccl

yurikadoi, 2019-01-05:
@lapishc 
I do not know if this is any helpful but I convert .continuous file (from OE) into .dat file  to use on JRCLUST.
The converter code is here: https://github.com/open-ephys/analysis-tools/blob/master/Python3/OpenEphys.py
Yurika

aliddell, 2019-02-28:
I just came across this for Matlab: https://github.com/cortex-lab/spikes/blob/master/preprocessing/oe2dat.m

lapishc, 2019-03-03:
Thanks Alan!

ccl

sandraromerop, 2019-08-08:
regarding this, does a .dat or .bin file with samples written as [nChannelsxnSamples] suffice to load into JRCLUST?

aliddell, 2019-08-12:
@sandraromerop Yes.