JaneliaSciComp/JRCLUST #84: Some sites are not ignored

LiuDaveLiu, 2019-02-08:
The .prm is the same as provided in the email.

If you found a bug, please provide **all** of the following:

#### Output of `ver`

#### Output of `gpuDevice`

#### The command you used
(e.g., `jrc detect-sort myparam.prm`)

jrc preview E:\MAP\ww2\dl56\20181120\dl56_g0_t100.imec.ap.prm

#### Console output

[image]

jamesjun, 2019-02-08:
You might be better off creating a new map containing the bottom half of
the probe Please run `jrc probe imec3_opt3_195_245.prb` or see the probe
file `probe\imec3_opt3_195_245.prb`

-James

jamesjun, 2019-02-08:
See how I subselected the channels on line 1.

https://github.com/JaneliaSciComp/JRCLUST/blob/master/probes/imec3_opt3_195_245.prb

-James

LiuDaveLiu, 2019-02-08:
This have not been an issue in v3.3.0 of JRCLust. I only noticed this in v4.

aliddell, 2019-02-10:
Yeah I must have taken this out. Can't imagine why. I'll add it back in.

aliddell, 2019-02-11:
Fixed in b742df7

closed 2019-02-11 by aliddell