JaneliaSciComp/JRCLUST #85: Manual curation is slow in v4

LiuDaveLiu, 2019-02-09:
Everything is much slower in the manual curation of v4 compared to v3. Merging, splitting, saving, etc.

If you have a usage or support question, come join our [Gitter](https://gitter.im/JRCLUST/community) and ask your question there.
If you found a bug, please provide **all** of the following:

#### Output of `ver`
Same as before
#### Output of `gpuDevice`
Same as before
#### The command you used
(e.g., `jrc detect-sort myparam.prm`)

#### If manually sorting, what it is you tried to do
Merging units is much slower than in v3.
Computing waveform correlation...	took 1.7s
Computing self correlation
	................................................................................................................................................................................................................................................................................................................................................................................................................
	took 4.9s
Calculating cluster quality...
	took 0.1s
Computing waveform correlation...	took 3.1s
Computing self correlation
	...............................................................................................................................................................................................................................................................................................................................................................................................................
	took 4.9s
Calculating cluster quality...
	took 0.0s
Computing waveform correlation...	took 0.2s
Computing self correlation
	..............................................................................................................................................................................................................................................................................................................................................................................................................
	took 4.9s
Calculating cluster quality...
	took 0.0s
Computing waveform correlation...	took 0.1s
Computing self correlation
	.............................................................................................................................................................................................................................................................................................................................................................................................................
	took 5.2s
Calculating cluster quality...
	took 0.1s
Computing waveform correlation...	took 3.9s
Computing self correlation
	............................................................................................................................................................................................................................................................................................................................................................................................................
	took 5.6s
Calculating cluster quality...
	took 0.0s
#### Console output

aliddell, 2019-02-10:
I'll need to profile this and see where the bottleneck is.

jamesjun, 2019-02-10:
Try running ‘jrc unit-test manual’ and it should create a profiler report. It should be compared between version 3 and 4 to determine what’s slowing down. 

-James

aliddell, 2019-02-11:
Will do

aliddell, 2019-02-19:
This is partially addressed in 4539c5f2bd5dc06b6743b7c1cfabbd0998075493. Auto-merge is still much slower, that's a work in progress.