JaneliaSciComp/JRCLUST #86: Error is v4 while saving

LiuDaveLiu, 2019-02-09:
Error while saving in manual curation

If you have a usage or support question, come join our [Gitter](https://gitter.im/JRCLUST/community) and ask your question there.
If you found a bug, please provide **all** of the following:

#### Output of `ver`
Same as before
#### Output of `gpuDevice`
Same as before
#### The command you used
(e.g., `jrc detect-sort myparam.prm`)
Same as before
#### If manually sorting, what it is you tried to do
Save
#### Console output
Error using jrclust.controllers.curate.CurateController/saveFiles
Too many input arguments.

Error in jrclust.controllers.curate.CurateController>@(varargin)obj.saveFiles(varargin{:}) (line
641)
            uimenu(obj.hMenus('FileMenu'), 'Label', 'Save', 'Callback', @obj.saveFiles);
 
Error while evaluating Menu Callback.

aliddell, 2019-02-10:
Were you at least able to save when you exited?

LiuDaveLiu, 2019-02-10:
Yes, I was able to save when I exited, so this is a low priority issue.

aliddell, 2019-02-11:
Fixed in b742df7

closed 2019-02-11 by aliddell