JaneliaSciComp/JRCLUST #87: Clusters on distant recording sites after splitting

Dario55, 2019-02-13:
If you have a usage or support question, come join our [Gitter](https://gitter.im/JRCLUST/community) and ask your question there.
If you found a bug, please provide **all** of the following:

#### Output of `ver`

----------------------------------------------------------------------------------------------------
MATLAB Version: 9.2.0.538062 (R2017a)
MATLAB License Number: 649021
Operating System: Microsoft Windows 7 Enterprise  Version 6.1 (Build 7601: Service Pack 1)
Java Version: Java 1.7.0_60-b19 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
----------------------------------------------------------------------------------------------------

#### Output of `gpuDevice`

ans = 

  CUDADevice with properties:

                      Name: 'GeForce GTX 1080 Ti'
                     Index: 1
         ComputeCapability: '6.1'
            SupportsDouble: 1
             DriverVersion: 9.2000
            ToolkitVersion: 8
        MaxThreadsPerBlock: 1024
          MaxShmemPerBlock: 49152
        MaxThreadBlockSize: [1024 1024 64]
               MaxGridSize: [2.1475e+09 65535 65535]
                 SIMDWidth: 32
               TotalMemory: 1.1811e+10
           AvailableMemory: 9.2075e+09
       MultiprocessorCount: 28
              ClockRateKHz: 1620000
               ComputeMode: 'Default'
      GPUOverlapsTransfers: 1
    KernelExecutionTimeout: 0
          CanMapHostMemory: 1
           DeviceSupported: 1
            DeviceSelected: 1

#### The command you used
 jrc3 spikesort G:\07122018_bck\07122018_364_1a_g0_t0.denoised.imec.ap_imec3_opt1.prm;
 jrc3 manual G:\07122018_bck\07122018_364_1a_g0_t0.denoised.imec.ap_imec3_opt1.prm;
#### If manually sorting, what it is you tried to do
splitting a cluster into two clusters
#### Console output

Hello,

I had an issue when splitting clusters. The example I report below is from jrc3, but we observed a similar behaviour in jrc2 as well. Here clusters 812 and 813 were previously part of same cluster, that I have manually split. I don't quite understand why the algorithm has merged waveform from recording sites that are so far apart. Is that correct? Would you please provide an intuition for why this happens? Thank you

[image]

[image]

aliddell, 2019-02-19:
This isn't due to merging. From the data you gave me I saw this phenomenon skipping waveform-based merging altogether. This looks like "nearest-neighbor-of-a-nearest-neighbor" but I need to do some more tests to be sure.

aliddell, 2019-04-17:
@Dario55 with the latest commit I'm no longer seeing this issue. Can you verify it works for you as well?

aliddell, 2019-04-23:
Closing as fixed in #110

closed 2019-04-23 by aliddell