JaneliaSciComp/JRCLUST #88: manual interface becomes unresponsive after cancelling auto-delete or auto-merge

jenniferColonell, 2019-02-15:
If you have a usage or support question, come join our [Gitter](https://gitter.im/JRCLUST/community) and ask your question there.
If you found a bug, please provide **all** of the following:

#### Output of `ver`
-----------------------------------------------------------------------------------------------------
MATLAB Version: 9.5.0.1033004 (R2018b) Update 2
MATLAB License Number: 822475
Operating System: Microsoft Windows 10 Pro Version 10.0 (Build 17134)
Java Version: Java 1.8.0_152-b16 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 9.5         (R2018b)
Curve Fitting Toolbox                                 Version 3.5.8       (R2018b)
Image Processing Toolbox                              Version 10.3        (R2018b)
Optimization Toolbox                                  Version 8.2         (R2018b)
Parallel Computing Toolbox                            Version 6.13        (R2018b)
Signal Processing Toolbox                             Version 8.1         (R2018b)
Statistics and Machine Learning Toolbox               Version 11.4        (R2018b)

#### Output of `gpuDevice`
 CUDADevice with properties:

                      Name: 'GeForce GTX 1050 Ti with Max-Q Design'
                     Index: 1
         ComputeCapability: '6.1'
            SupportsDouble: 1
             DriverVersion: 10
            ToolkitVersion: 9.1000
        MaxThreadsPerBlock: 1024
          MaxShmemPerBlock: 49152
        MaxThreadBlockSize: [1024 1024 64]
               MaxGridSize: [2.1475e+09 65535 65535]
                 SIMDWidth: 32
               TotalMemory: 4.2950e+09
           AvailableMemory: 3.4351e+09
       MultiprocessorCount: 6
              ClockRateKHz: 1290500
               ComputeMode: 'Default'
      GPUOverlapsTransfers: 1
    KernelExecutionTimeout: 1
          CanMapHostMemory: 1
           DeviceSupported: 1
            DeviceSelected: 1
#### The command you used
(e.g., `jrc detect-sort myparam.prm`)
jrc manual param.prm
#### If manually sorting, what it is you tried to do

When trying to use Edit->Merge Auto or Edit->Delete Auto, I find that if I cancel the dialog, all of the other figures stop accepting mouse clicks, nor can I change the View in the Averaged waveform window.

Also, Edit->Delete Auto does not delete the clusters in the display.

Edit->Delete Auto does not change any spike assignments in the res.mat file.
#### Console output

aliddell, 2019-02-15:
Thanks for this. One thing that isn't clear from the docs is that `spikeClusters` in the `_res.mat` file is the initial clustering and modifications are stored in the `hClust` object. Everything else is worth looking into. Mind posting the git commit hash?

jenniferColonell, 2019-02-15:
Thanks, Alan. I checked the assignments in the hClust object -- in my test, they did not change after an auto-delete. 
I'm running JRCLUST-4.0.0-alpha.4, which has git commit hash b742df7.

aliddell, 2019-02-19:
This should be fixed in the [latest patch](https://github.com/JaneliaSciComp/JRCLUST/releases/tag/v4.0.0-alpha.5). Thanks for the report!

closed 2019-02-19 by aliddell