JaneliaSciComp/JRCLUST #92: Where did gridSorter disappear to?

NoahDolev, 2019-02-28:
Hi,

In previous versions of JRClust there was an import and export gridSorter method that has disappeared from the code base. 

```{Matlab}
function export_gridSorter_(varargin)
% export_csv_(hObject, event)
% if nargin<2, 
fZeroIndex = 0; %zero-based index export (disable to export as matlab index starting from 1)

% S0 = get(0, 'UserData');
if nargin==2
    [S0, P, S_clu] = get0_();
elseif nargin==1
    P = varargin{1};
    vcFile_prm = P.vcFile_prm;
    S0 = load_cached_(P, 0);
    if isempty(S0), fprintf(2, 'Cannot find _jrc.mat.\n'); return; end %exit if file doesn't exist
    P = S0.P;
end
...
```
Do I have to re-implement it or is there some replacement for it? 

Thanks,
Noah

NoahDolev, 2019-02-28:
Hi,

I discovered the issue. It was implemented by a third party. :) Closing the issue.  Sorry.

Best,
Noah

closed 2019-02-28 by NoahDolev

aliddell, 2019-02-28:
Hi Noah,

I'm not familiar with gridSorter. Can you point me to a paper? Also, if there are no licensing collisions, I'd love the contribution!