JaneliaSciComp/JRCLUST #93: v4 bugs in manual curation

WeissShahaf, 2019-03-03:
i'm using trying version: JRCLUST v4.0.0-alpha-patch5 "Edward", while sorting works well and very fast,
I cannot use manual curation split or merge functionalities.  After trying either, I get "operation failed" message. After which i cannot select a different unit or continue to do anything else. Even a "close all" command results in errors (see attached files for errors).
 
This is replicated on both a Desktop dual socket, 8 cor xeon CPUs + P5000 and a laptop with i6700HQ + GTX1060, using Matlab 2018a, and 2013a, on windows 10. 

If you found a bug, please provide **all** of the following:

#### Output of `ver`
>> ver
-----------------------------------------------------------------------------------------------------
MATLAB Version: 9.4.0.813654 (R2018a)
MATLAB License Number: 40523909
Operating System: Microsoft Windows 10 Home Version 10.0 (Build 17134)
Java Version: Java 1.8.0_144-b01 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode

-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 9.4         (R2018a)
Simulink                                              Version 9.1         (R2018a)
Aerospace Blockset                                    Version 3.21        (R2018a)
Aerospace Toolbox                                     Version 2.21        (R2018a)
Antenna Toolbox                                       Version 3.1         (R2018a)
Audio System Toolbox                                  Version 1.4         (R2018a)
Automated Driving System Toolbox                      Version 1.2         (R2018a)
Bioinformatics Toolbox                                Version 4.10        (R2018a)
Communications System Toolbox                         Version 6.6         (R2018a)
Computer Vision System Toolbox                        Version 8.1         (R2018a)
Control System Toolbox                                Version 10.4        (R2018a)
Curve Fitting Toolbox                                 Version 3.5.7       (R2018a)
DSP System Toolbox                                    Version 9.6         (R2018a)
Data Acquisition Toolbox                              Version 3.13        (R2018a)
Database Toolbox                                      Version 8.1         (R2018a)
Datafeed Toolbox                                      Version 5.7         (R2018a)
Econometrics Toolbox                                  Version 5.0         (R2018a)
Embedded Coder                                        Version 7.0         (R2018a)
Filter Design HDL Coder                               Version 3.1.3       (R2018a)
Financial Instruments Toolbox                         Version 2.7         (R2018a)
Financial Toolbox                                     Version 5.11        (R2018a)
Fixed-Point Designer                                  Version 6.1         (R2018a)
Fuzzy Logic Toolbox                                   Version 2.3.1       (R2018a)
GPU Coder                                             Version 1.1         (R2018a)
Global Optimization Toolbox                           Version 3.4.4       (R2018a)
HDL Coder                                             Version 3.12        (R2018a)
HDL Verifier                                          Version 5.4         (R2018a)
Image Acquisition Toolbox                             Version 5.4         (R2018a)
Image Processing Toolbox                              Version 10.2        (R2018a)
Instrument Control Toolbox                            Version 3.13        (R2018a)
LTE HDL Toolbox                                       Version 1.1         (R2018a)
LTE System Toolbox                                    Version 2.6         (R2018a)
MATLAB Coder                                          Version 4.0         (R2018a)
MATLAB Compiler                                       Version 6.6         (R2018a)
MATLAB Compiler SDK                                   Version 6.5         (R2018a)
MATLAB Report Generator                               Version 5.4         (R2018a)
Mapping Toolbox                                       Version 4.6         (R2018a)
Model Predictive Control Toolbox                      Version 6.1         (R2018a)
Model-Based Calibration Toolbox                       Version 5.4         (R2018a)
Neural Network Toolbox                                Version 11.1        (R2018a)
OPC Toolbox                                           Version 4.0.5       (R2018a)
Optimization Toolbox                                  Version 8.1         (R2018a)
Parallel Computing Toolbox                            Version 6.12        (R2018a)
Partial Differential Equation Toolbox                 Version 3.0         (R2018a)
Phased Array System Toolbox                           Version 3.6         (R2018a)
Polyspace Bug Finder                                  Version 2.5         (R2018a)
Polyspace Code Prover                                 Version 9.9         (R2018a)
Powertrain Blockset                                   Version 1.3         (R2018a)
Predictive Maintenance Toolbox                        Version 1.0         (R2018a)
RF Blockset                                           Version 7.0         (R2018a)
RF Toolbox                                            Version 3.4         (R2018a)
Risk Management Toolbox                               Version 1.3         (R2018a)
Robotics System Toolbox                               Version 2.0         (R2018a)
Robust Control Toolbox                                Version 6.4.1       (R2018a)
[... 50 lines cut ...]
               MaxGridSize: [2.1475e+09 65535 65535]
                 SIMDWidth: 32
               TotalMemory: 6.4425e+09
           AvailableMemory: 5.2310e+09
       MultiprocessorCount: 10
              ClockRateKHz: 1670500
               ComputeMode: 'Default'
      GPUOverlapsTransfers: 1
    KernelExecutionTimeout: 0
          CanMapHostMemory: 1
           DeviceSupported: 1
            DeviceSelected: 1
>> 
#### The command you used
jrc manual CSC_Data.prm  
jrc full CSC_Data.prm
#### If manually sorting, what it is you tried to do
spliiting or merging

#### Console output
[post_automerge.txt](https://github.com/JaneliaSciComp/JRCLUST/files/2923600/post_automerge.txt)
[post_split_selecting_a_unit.txt](https://github.com/JaneliaSciComp/JRCLUST/files/2923601/post_split_selecting_a_unit.txt)
[split.txt](https://github.com/JaneliaSciComp/JRCLUST/files/2923602/split.txt)
[automerge.txt](https://github.com/JaneliaSciComp/JRCLUST/files/2923603/automerge.txt)
[close all.txt](https://github.com/JaneliaSciComp/JRCLUST/files/2923610/close.all.txt)

aliddell, 2019-03-04:
Did you download the release or did you clone it? If you cloned it, can you post the commit hash for me?

aliddell, 2019-03-04:
Also, could you inspect your `_res.mat` file and tell me what the dimension of `siteThresh` is, and whether or not you have a field called `meanSiteThresh`? Thanks!

WeissShahaf, 2019-03-04:
I cloned the latest release. 
commit a567eac647beb98e421762fda8f45a2c09e86707.

the recording is from 64 channels.
size(siteThresh) is  64    57
meanSiteThresh exist and  size( meanSiteThresh) is    64     1

aliddell, 2019-03-04:
Unfortunately I can't reproduce your problem with any of the data I have. This little script should replicate where the main failure is. Can you run it line by line and let me know where it fails?

```matlab
load('CSC_Data_res.mat', 'hClust', 'meanSiteThresh');
hClust.computeMeanWaveforms();
unitVmin = squeeze(min(hClust.meanWfGlobal));
unitPeaks_ = jrclust.utils.rowColSelect(unitVmin, hClust.clusterSites, 1:hClust.nClusters);
siteRMS_ = jrclust.utils.bit2uV(single(meanSiteThresh(:))/hClust.hCfg.qqFactor, hClust.hCfg);
unitSNR_ = abs(unitPeaks_)./siteRMS_(hClust.clusterSites);
nSitesOverThresh_ = sum(bsxfun(@lt, unitVmin, - siteRMS_*hClust.hCfg.qqFactor), 1)';
```

WeissShahaf, 2019-03-04:
it all ran without failing.

WeissShahaf, 2019-03-04:
i've put the all data files in a dropbox folder:
https://www.dropbox.com/sh/icnma1r4hh4dujc/AACjo-W06Y6NtUBexx2ADvrYa?dl=0

aliddell, 2019-03-05:
Pull the latest and give it a whirl. You shouldn't see this issue anymore.

WeissShahaf, 2019-03-05:
Thanks!
that solved the issue

aliddell, 2019-03-05:
Awesome! Thanks for the bug report.