JaneliaSciComp/JRCLUST #103: Error using jsondecode while sorting (release 4a6)

oliriley, 2019-03-19:
I am encountering issues with sorting detected spikes on the most recent release. Detection itself appears to be functional, and spikes are displayed when using `jrc preview`.

#### Output of `ver`
MATLAB Version: 9.5.0.944444 (R2018b)
MATLAB License Number: 703789
Operating System: Microsoft Windows 10 Home Version 10.0 (Build 17134)
Java Version: Java 1.8.0_152-b16 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode

#### Output of `gpuDevice`
CUDADevice with properties:
                      Name: 'GeForce GTX 1070'
                     Index: 1
         ComputeCapability: '6.1'
            SupportsDouble: 1
             DriverVersion: 10
            ToolkitVersion: 9.1000
        MaxThreadsPerBlock: 1024
          MaxShmemPerBlock: 49152
        MaxThreadBlockSize: [1024 1024 64]
               MaxGridSize: [2.1475e+09 65535 65535]
                 SIMDWidth: 32
               TotalMemory: 8.5899e+09
           AvailableMemory: 6.9873e+09
       MultiprocessorCount: 16
              ClockRateKHz: 1695000
               ComputeMode: 'Default'
      GPUOverlapsTransfers: 1
    KernelExecutionTimeout: 1
          CanMapHostMemory: 1
           DeviceSupported: 1
            DeviceSelected: 1

#### The command you used
I previously ran:
`jrc detect cullen_lab_2_009_g0_t0.imec0.ap.prm`

which created the expected output files without issue. I then ran:
`jrc sort cullen_lab_2_009_g0_t0.imec0.ap.prm`

#### Console output
2019-03-19 13:20:43 Loading C:\Users\Oli\Documents\MATLAB\JRCLUST 4a6\cullen_lab_2_009_g0_t0.imec0.ap_res.mat...
2019-03-19 13:20:43 Finished loading C:\Users\Oli\Documents\MATLAB\JRCLUST 4a6\cullen_lab_2_009_g0_t0.imec0.ap_res.mat (took 0.03 s)
2019-03-19 13:20:43 Loading C:\Users\Oli\Documents\MATLAB\JRCLUST 4a6\cullen_lab_2_009_g0_t0.imec0.ap_raw.jrc...
2019-03-19 13:20:43 Finished loading C:\Users\Oli\Documents\MATLAB\JRCLUST 4a6\cullen_lab_2_009_g0_t0.imec0.ap_raw.jrc (took 0.03 s)
2019-03-19 13:20:43 Loading C:\Users\Oli\Documents\MATLAB\JRCLUST 4a6\cullen_lab_2_009_g0_t0.imec0.ap_filt.jrc...
2019-03-19 13:20:43 Finished loading C:\Users\Oli\Documents\MATLAB\JRCLUST 4a6\cullen_lab_2_009_g0_t0.imec0.ap_filt.jrc (took 0.02 s)
2019-03-19 13:20:43 Loading C:\Users\Oli\Documents\MATLAB\JRCLUST 4a6\cullen_lab_2_009_g0_t0.imec0.ap_features.jrc...
2019-03-19 13:20:43 Finished loading C:\Users\Oli\Documents\MATLAB\JRCLUST 4a6\cullen_lab_2_009_g0_t0.imec0.ap_features.jrc (took 0.00 s)
2019-03-19 13:20:43 Using spikes detected on 19-Mar-2019 13:16:41
2019-03-19 13:20:43 Clearing GPU memory...
2019-03-19 13:20:44 GPU memory cleared (took 1.12 s)
2019-03-19 13:20:44 Sorting detected spikes...
2019-03-19 13:20:44 Computing rho...
2019-03-19 13:20:45 Site 2: rho cutoff, 59512.00; average rho: 0.00013 (8 spikes)
...[continues for many sites]
2019-03-19 13:20:46 Site 203: rho cutoff, NaN; average rho: 0.03430 (1 spikes)
2019-03-19 13:20:46 Finished computing rho (took 1.69 s)
2019-03-19 13:20:46 Computing delta...
2019-03-19 13:20:46 Site 2: median delta: 1.44206 (8 spikes)
...[continues for many sites]
2019-03-19 13:20:46 Site 203: median delta: NaN (1 spikes)
2019-03-19 13:20:46 Finished computing delta (took 0.28 s)
2019-03-19 13:20:46 Assigning clusters (nClusters: 1024)...
2019-03-19 13:20:46 iter 1, 23 clusters remain unassigned
...[continues for each iteration]
2019-03-19 13:20:46 iter 10, 781 clusters remain unassigned
2019-03-19 13:20:46 Finished assigning clusters (was 0, now 86: 938 clusters with fewer than 30 spikes removed) (took 0.11 s)

**Error using jsondecode**
**JSON text must be a character vector or a scalar non-missing string.**

**Error in jrclust.sort.DensityPeakClustering (line 74)**
            dpFields = jsondecode(fread(fid, inf, '*char'));

**Error in jrclust.sort.SortController/sort (line 53)**
            hClust = jrclust.sort.DensityPeakClustering(res, dRes, obj.hCfg);

**Error in JRC/sort (line 34)**
    sRes = obj.hSort.sort(obj.res);

**Error in JRC/run (line 20)**
        obj.sort();

**Error in jrc (line 20)**
        hJRC.run();

aliddell, 2019-03-19:
In your repository you should find a file at json/DensityPeakClustering.json. Have you made any changes to it? If not, can you post the output of

```matlab
fread(open(fullfile(jrclust.utils.basedir(), 'json', 'DensityPeakClustering.json')))
```

for me?

oliriley, 2019-03-19:
I do see that file, and I have not edited it. For reference, the entirety of DensityPeakClustering.json as it exists in my JRCLUST folder is as follows:
`{
    "vector fields": [
        "clusterCenters",
        "nSitesOverThresh",
        "unitSNR"
    ]
}`

Output of `fread(open(fullfile(jrclust.utils.basedir(), 'json', 'DensityPeakClustering.json')))` throws an invalid file identifier error. If I use `fopen` instead of `open`, I get the following:

ans =
   123
    10
    32
    32
    32
    32
    34
   118
   101
    99
   116
   111
   114
    32
   102
   105
   101
   108
   100
   115
    34
    58
    32
    91
    10
    32
    32
    32
    32
    32
    32
    32
    32
    34
    99
   108
   117
   115
   116
   101
   114
    67
   101
   110
   116
   101
   114
   115
    34
    44
    10
    32
    32
    32
    32
    32
    32
    32
    32
    34
   110
    83
   105
   116
   101
   115
    79
   118
   101
   114
    84
   104
   114
   101
   115
   104
    34
    44
    10
    32
    32
    32
    32
    32
    32
    32
    32
    34
   117
   110
   105
   116
    83
    78
    82
    34
    10
    32
    32
    32
    32
    93
    10
   125

aliddell, 2019-03-19:
Sorry, I meant

```matlab
fread(fopen(fullfile(jrclust.utils.basedir(), 'json', 'DensityPeakClustering.json')), '*char')
```

oliriley, 2019-03-19:
That gives a character-by-character version of what I posted above:
`{ "vector fields": [ "clusterCenters", "nSitesOverThresh", "unitSNR" ] }`

In full, it gives:
'{'
    '↵'
    ' '
    ' '
    ' '
    ' '
    '"'
    'v'
    'e'
    'c'
    't'
    'o'
    'r'
    ' '
    'f'
    'i'
    'e'
    'l'
    'd'
    's'
    '"'
    ':'
    ' '
    '['
    '↵'
    ' '
    ' '
    ' '
    ' '
    ' '
    ' '
    ' '
    ' '
    '"'
    'c'
    'l'
    'u'
    's'
    't'
    'e'
    'r'
    'C'
    'e'
    'n'
    't'
    'e'
    'r'
    's'
    '"'
    ','
    '↵'
    ' '
    ' '
    ' '
    ' '
    ' '
    ' '
    ' '
    ' '
    '"'
    'n'
    'S'
    'i'
    't'
    'e'
    's'
    'O'
    'v'
    'e'
    'r'
    'T'
    'h'
    'r'
    'e'
    's'
    'h'
    '"'
    ','
    '↵'
    ' '
    ' '
    ' '
    ' '
    ' '
    ' '
    ' '
    ' '
    '"'
    'u'
    'n'
    'i'
    't'
    'S'
    'N'
    'R'
    '"'
    '↵'
    ' '
    ' '
    ' '
    ' '
    ']'
    '↵'
    '}'

aliddell, 2019-03-19:
This must be 2018 being fussy. I can replicate on 2018b but not on 2017b. I'll have a patch up in a few minutes. Thanks for catching it.

closed 2019-03-19 by aliddell

oliriley, 2019-03-19:
Thanks so much for the quick response! I will test the patch ASAP

oliriley, 2019-03-19:
@aliddell, works locally now

commit 10cf9a15 (2019-03-19): fix #103
commit 1cee5616 (2019-03-19): Merge pull request #104 from vidriotech/master

fix #103