JaneliaSciComp/JRCLUST #108: 'A field named 'hClust' doesn't exist' using 4.0.0-beta

oliriley, 2019-04-02:
May be related to issue [102](https://github.com/JaneliaSciComp/JRCLUST/issues/102)?

#### Output of `ver`
MATLAB Version: 9.5.0.944444 (R2018b)
Operating System: Microsoft Windows 10 Home Version 10.0 (Build 17134)
Java Version: Java 1.8.0_152-b16 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode

#### The command you used
After unzipping 4-beta release and using `bootstrap`, try `detect-sort` with all default params.
`jrc detect-sort cullen_lab_2_009_g0_t0.imec0.ap.prm`

#### Console output
`bootstrap` seems to work fine, and all `detect-sort` outputappears normal until the final part (saving to results mat file):
...
`2019-04-01 20:34:57 Saving results to C:\Users\Oli\Documents\MATLAB\JRCLUST-4-beta\cullen_lab_2_009_g0_t0.imec0.ap_res.mat...`
`Error using rmfield (line 65)`
`A field named 'hClust' doesn't exist.`

`Error in JRC/saveRes (line 24)`
    `res_ = rmfield(obj.res, 'hClust');`

`Error in JRC/detect (line 27)`
    `obj.saveRes(0);`

`Error in JRC/run (line 13)`
        `obj.detect();`

`Error in jrc (line 20)`
        `hJRC.run();`

aliddell, 2019-04-02:
Thanks for catching this.

closed 2019-04-02 by aliddell

oliriley, 2019-04-02:
@aliddell I believe there also needs to be one more line removed; line 78 in saveRes.m has

obj.res.hClust = hClust;

...which should now happen inside the conditional earlier in the function. Leaving the line in throws 'Undefined function or variable 'hClust'.', while commenting out this line appears to fix that issue.

aliddell, 2019-04-02:
d'oh thanks

aliddell, 2019-04-02:
@oliriley Okay, fixed now. That line was useless anyway.

commit f21014bc (2019-04-02): Fix #108