JaneliaSciComp/JRCLUST #110: Multiple errors in displaying results of 'jrc manual

oliriley, 2019-04-03:
#### Output of `ver`

#### Output of `gpuDevice`

#### The command you used
`jrc bootstrap`
`jrc detect-sort myparam.prm` (appears to finish normally
`jrc manual myparam.prm`
Keypress `s`
Select a clustering method and click `OK`
Click `Keep Splits`

#### If manually sorting, what it is you tried to do
Split a cluster using 's' key, using any of the provided clustering algorithms.

#### Console output
A dialogue box appears reading "Operation failed", with an 'okay' button.

In the console:
Warning: Failed to split: Index exceeds the number of array elements (2234). 
> In jrclust.curate.CurateController/splitCluster (line 43)
  In jrclust.curate.CurateController/autoSplit (line 183)
  In jrclust.curate.CurateController/keyPressFigWav (line 137)
  In jrclust.curate.CurateController/plotAllFigures>@(varargin)obj.keyPressFigWav(varargin{:}) (line 40)

This seems to *not* happen if I am splitting a cluster that was created immediately beforehand by merging two previously existing clusters.

aliddell, 2019-04-03:
I've seen the first error when clusters have spikes from all over the probe. Can you load your spikeSites and spikesByCluster fields from your res.mat file and do

`cellfun(@(spikes) numel(unique(spikeSites(spikes))), spikesByCluster)`

for me?

For the second error, can you tell me if this happens if you split one cluster into 3 or more clusters?

oliriley, 2019-04-03:
cellfun(@(spikes) numel(unique(spikeSites(spikes))), spikesByCluster):  
  Columns 1 through 23
   237     3     1     1     2     2     2     5     8     2     2     4     4     9     8     4     2     5     2     2     2    10     5

  Columns 24 through 46
     3     5     4     2     2     8     2     4     4     5     3     1     5     3     4     7     5     1     4     4     4     3     4

  Columns 47 through 69
     8     5     5     4     4     3     5     3     2     2     5     4     3     3     2     5     2     2     1     1     1     1     2

  Columns 70 through 92
     2     2     2     1     4     1     2     1     3     3     1     2     1     2     2     3     1     2     4     1     2     1     3

  Columns 93 through 115
     5     2     2     3     1     1     3     2     1     4     3     3     2     2     2     3     3     3     6     5     3     1     2

  Columns 116 through 138
     3     1     1     2     4     4     3     5     3     6     3     7     3     5     3     3     2     3     2     7     5     2     1

  Columns 139 through 161
     4     4     2     3     2     2     6     1     2     4     5     5     3     4     7     4     4     5     1     2     3     1     4

  Columns 162 through 184
     3     4     3     2     1     4     4     1     2     3     3     3     4     7     4     4     3     5     3     2     1     7     3

  Columns 185 through 192
     4     5     2     2     2     3     1     1

oliriley, 2019-04-03:
3-Splitting appears to fail in the same cases as 2-splitting; if I try to split a cluster 3 ways, and get an operation failure, it will also fail to split 2 ways, and vice versa.

aliddell, 2019-04-03:
> cellfun(@(spikes) numel(unique(spikeSites(spikes))), spikesByCluster):
> Columns 1 through 23
> 237 3 1 1 2 2 2 5 8 2 2 4 4 9 8 4 2 5 2 2 2 10 5
>
> Columns 24 through 46
> 3 5 4 2 2 8 2 4 4 5 3 1 5 3 4 7 5 [...]

Yeah, that first cluster is enormous and should not have been merged. Any chance you can send me this data so I can see why that might have happened? Fixing the first problem might address the second.

oliriley, 2019-04-03:
> Any chance you can send me this data so I can see why that might have happened?
Sure. It should be in your inbox shortly.

aliddell, 2019-04-04:
@oliriley I think I've addressed at least part of the problem you're seeing. Can you pull the latest and run

`jrc sort /path/to/params.prm`

and then try manually curating?

oliriley, 2019-04-05:
Will give it a try!

oliriley, 2019-04-05:
@aliddell you were right that it gives the same issues with giant first cluster (237 sites still) but the initial clusters are arranged by site number, and splitting works fine. Can post screenshots/console logs if it would be helpful.

Something to note: the the giant cluster-one exists even though its waveform similarity is <0.75, way below the minimum that should be allowed (I believe default is 0.98?). That makes me suspect that whatever is causing all the merges may be related to a missing check on similarity score.

aliddell, 2019-04-08:
@oliriley These spikes are all assigned to the same cluster based on feature similarity, before any waveform-based merging occurs. It's the same issue as #87, actually. It's going to take some careful thought to come up with a good solution to it.

aliddell, 2019-04-17:
@oliriley with the latest commit I'm no longer seeing this issue. Can you verify it works for you as well?

oliriley, 2019-04-22:
I'm testing now. It seems to have cleared up the first-large-cluster issue nicely!

aliddell, 2019-04-23:
Excellent! Good to close?

closed 2019-04-23 by oliriley