JaneliaSciComp/JRCLUST #115: refine cluster using polygon

agbondy, 2019-06-01:
In the feature view, is it possible to draw a polygon for the purposes of refining a cluster -- that is, to define the cluster as only those points within the polygon?

aliddell, 2019-06-03:
One can split the cluster in the feature view (press 'S' when the feature window is in the foreground), and from there you can decide what to do with the other spikes (keep, delete).

agbondy, 2019-06-03:
Sorry, I should have been more clear. What I mean is best illustrated by this example panel from the feature window. What you see here is that JRClust has identified a real unit but it has missed most of its events (i.e. grey background dots under the black cluster). What I'd like to do is redefine the unit by drawing a polygon to include all these events.  Is there a way to achieve this? I found this basic issue (JRClust assigning many of the units in a cluster to background) to be quite common.

[image]

aliddell, 2019-06-03:
No, this isn't possible in JRCLUST. The only way to approximate this is by finding (by trial and error, unfortunately) which unit or units these spikes belong to and merging them with the unit you've got selected.

agbondy, 2019-06-05:
This would be a very useful feature imo!