JaneliaSciComp/JRCLUST #117: incorrect parsing of 3b .meta file

agbondy, 2019-06-05:
In loadMetadata.m, because 3b probes create .meta files with a field ImSampRate, the condition on line 36 is satisfied and the ensuing code sets 10 reference channels and does other stuff that looks like the operating assumption is a 3a, not 3b, probe.

aliddell, 2019-06-05:
You're right, this should be updated to prefer 3B probes.

aliddell, 2019-06-06:
Can you give the latest commit a pull and let me know if that works for you?

agbondy, 2019-06-06:
Yup this runs just fine! Those changes will make starting up a neuropixels sort more streamlined, thanks!

However, I'm still confused why a .prb file is necessary at all when working with neuropixels data. Doesn't the .meta file have all the necessary information?

aliddell, 2019-06-07:
That's a good question. It boils down to legacy workflow and support for missing metadata.

We already have tested code for loading probe configurations from .prb files, so we may as well use it. It's simpler and less error-prone to write up a short probe file than it is to add more parsing logic.

Also, in the usual case we do have everything we need, but we want to accommodate datasets that don't have meta files, or have incomplete meta files. See, e.g., [this one](http://data.cortexlab.net/singlePhase3/), which has an LF meta file and no ~snsShankMap field.

closed 2019-08-01 by aliddell