JaneliaSciComp/JRCLUST #121: splitting high firing rate cells can take a very long time

agbondy, 2019-06-18:
On the order of ten minutes if using the "hierarchical clustering" option. Looking at the computer's resources, the GPU is idle and the CPU is only 10% active, suggesting it may be possible to dramatically improve performance of whatever the expensive underlying operation is?

aliddell, 2019-07-03:
@agbondy sorry for the delay on this. I have a user locally who started seeing this or something similar right around the time you did. Some questions:

1. What model CPU/how much memory do you have?
2. What size clusters are you looking at?
3. Are you seeing this with all clusters?
4. If you set a breakpoint [here](https://github.com/JaneliaSciComp/JRCLUST/blob/master/%2Bjrclust/%2Bcurate/%40CurateController/autoSplit.m#L184):
a) what size is `sampledSpikes`?
b) if you step through into `pca` (which is where I'm presuming this is happening, since it's happening there for us) and into its local functions, is it locking up at the call to `svd`?

We're seeing the svd taking unreasonably long for smallish (~1000x250) matrices here.

aliddell, 2019-07-26:
Would you mind pulling the latest version and seeing if this problem still presents itself?

agbondy, 2019-08-22:
Sorry for the super delayed reply. Was writing a grant. I'll pull the latest version and get back to you.

agbondy, 2019-08-22:
OK, updated to the latest code and it's still happening. It's happening in the hierarchical clustering function itself (matlab's linkage.m function) at line 257: " Z = linkagemex(Y',method,{'euc'},memEff); ". The size of Y is 140449 by 5, where 140449 is the number of events in the cluster. For this cluster, that line alone takes several minutes to run.  

I'm running Matlab R2018a a Dell Workstation with an Intel Xeon Gold 6136 CPU (12 cores at 3GHz), 384GB of DDR4 RAM and an NVIDIA Quadro P5000 GPU with 16GB of RAM. Matlab R2018b.