JaneliaSciComp/JRCLUST #123: bug- if no spikes in one chunk

WeissShahaf, 2019-07-04:
when detecting spikes, if the current chunk does not detect any spikes- jrclust crushes instead of going to the next data chunk.

this is sometimes solved by manually chunking bigger time bins, but then if another chunk contains too many spikes- it might go out of memory. also time consuming to fiddle with finding the right size if dataset is big (15 hours of recording time)

If you have a usage or support question, come join our [Gitter](https://gitter.im/JRCLUST/community) and ask your question there.
If you found a bug, please provide **all** of the following:

#### Output of `ver`
K>> ver
-----------------------------------------------------------------------------------------------------
MATLAB Version: 9.4.0.813654 (R2018a)
MATLAB License Number: 40523909
Operating System: Microsoft Windows 10 Enterprise Version 10.0 (Build 16299)
Java Version: Java 1.8.0_144-b01 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 9.4         (R2018a)
Simulink                                              Version 9.1         (R2018a)
Aerospace Blockset                                    Version 3.21        (R2018a)
Aerospace Toolbox                                     Version 2.21        (R2018a)
Antenna Toolbox                                       Version 3.1         (R2018a)
Audio System Toolbox                                  Version 1.4         (R2018a)
Automated Driving System Toolbox                      Version 1.2         (R2018a)
Bioinformatics Toolbox                                Version 4.10        (R2018a)
Communications System Toolbox                         Version 6.6         (R2018a)
Computer Vision System Toolbox                        Version 8.1         (R2018a)
Control System Toolbox                                Version 10.4        (R2018a)
Curve Fitting Toolbox                                 Version 3.5.7       (R2018a)
DSP System Toolbox                                    Version 9.6         (R2018a)
Data Acquisition Toolbox                              Version 3.13        (R2018a)
Database Toolbox                                      Version 8.1         (R2018a)
Datafeed Toolbox                                      Version 5.7         (R2018a)
Econometrics Toolbox                                  Version 5.0         (R2018a)
Embedded Coder                                        Version 7.0         (R2018a)
Filter Design HDL Coder                               Version 3.1.3       (R2018a)
Financial Instruments Toolbox                         Version 2.7         (R2018a)
Financial Toolbox                                     Version 5.11        (R2018a)
Fixed-Point Designer                                  Version 6.1         (R2018a)
Fuzzy Logic Toolbox                                   Version 2.3.1       (R2018a)
GPU Coder                                             Version 1.1         (R2018a)
GUI Layout Toolbox                                    Version 2.3.4       (R2018b)
GUI Layout Toolbox                                    Version 1.17        (R2013b)
GUI Layout Toolbox                                    Version 1.17        (R2013b)
Global Optimization Toolbox                           Version 3.4.4       (R2018a)
HDL Coder                                             Version 3.12        (R2018a)
HDL Verifier                                          Version 5.4         (R2018a)
Image Acquisition Toolbox                             Version 5.4         (R2018a)
Image Processing Toolbox                              Version 10.2        (R2018a)
Instrument Control Toolbox                            Version 3.13        (R2018a)
LTE HDL Toolbox                                       Version 1.1         (R2018a)
LTE System Toolbox                                    Version 2.6         (R2018a)
MATLAB Coder                                          Version 4.0         (R2018a)
MATLAB Compiler                                       Version 6.6         (R2018a)
MATLAB Compiler SDK                                   Version 6.5         (R2018a)
MATLAB Report Generator                               Version 5.4         (R2018a)
Mapping Toolbox                                       Version 4.6         (R2018a)
Model Predictive Control Toolbox                      Version 6.1         (R2018a)
Model-Based Calibration Toolbox                       Version 5.4         (R2018a)
Neural Network Toolbox                                Version 11.1        (R2018a)
OPC Toolbox                                           Version 4.0.5       (R2018a)
Optimization Toolbox                                  Version 8.1         (R2018a)
Parallel Computing Toolbox                            Version 6.12        (R2018a)
Partial Differential Equation Toolbox                 Version 3.0         (R2018a)
Phased Array System Toolbox                           Version 3.6         (R2018a)
Polyspace Bug Finder                                  Version 2.5         (R2018a)
Polyspace Code Prover                                 Version 9.9         (R2018a)
Powertrain Blockset                                   Version 1.3         (R2018a)
Predictive Maintenance Toolbox                        Version 1.0         (R2018a)
RF Blockset                                           Version 7.0         (R2018a)
RF Toolbox                                            Version 3.4         (R2018a)
Risk Management Toolbox                               Version 1.3         (R2018a)
[... 466 lines cut ...]
2019-07-04 13:56:43 Detected 0 spikes on site 381
2019-07-04 13:56:43 Detected 0 spikes on site 382
2019-07-04 13:56:43 Detected 0 spikes on site 383
2019-07-04 13:56:43 Finished detecting spikes (took 2.95 s)
2019-07-04 13:56:44 Merging duplicate spiking events...
2019-07-04 13:56:44 0 spiking events found (took 0.69 s)
Index in position 2 exceeds array bounds (must not exceed 1).
Error in jrclust.detect.DetectController/findSecondaryPeaks (line 7)
    spikeWindows2 = spikeWindows(:, evtNeighbors, :);
Error in jrclust.detect.DetectController/samplesToWindows (line 57)
        [spikeSites2, spikeSites3] = obj.findSecondaryPeaks(spikesFilt,
        spikeSites);
Error in jrclust.detect.DetectController/detectOneRecording (line 194)
        loadData = obj.samplesToWindows(loadData);
Error in jrclust.detect.DetectController/detect (line 66)
        recData = obj.detectOneRecording(hRec, [rawFid, filtFid], impTimes,
        impSites, siteThresh);
Error in JRC/detect (line 16)
    dRes = obj.hDetect.detect();
Error in JRC/run (line 13)
        obj.detect();
Error in jrc (line 20)
        hJRC.run(); 
7       spikeWindows2 = spikeWindows(:, evtNeighbors, :);
K>>

WeissShahaf, 2019-08-01:
is there a fix for this issue?

aliddell, 2019-08-01:
@WeissShahaf you're right, I dropped the ball here. There is one now.

aliddell, 2019-08-29:
@WeissShahaf can you let me know if you're still seeing this problem?

aliddell, 2019-09-03:
I'm going to tentatively close this. If you see it again, feel free to reopen.

closed 2019-09-03 by aliddell