JaneliaSciComp/JRCLUST #124: bug/feature in splitting

thomasslai, 2019-07-23:
when splitting clusters, the mean waveform displayed is that of the lowest channel/site number in the cluster, when it should be the max site. this leads to some questionable splitting when the lowest channel is not picking up the full waveform

If you have a usage or support question, come join our [Gitter](https://gitter.im/JRCLUST/community) and ask your question there.
If you found a bug, please provide **all** of the following:

JRCLUST v4.0.0-beta-p1 "Edward"
#### Output of `ver`
>> ver
-----------------------------------------------------------------------------------------------------
MATLAB Version: 9.3.0.713579 (R2017b)
MATLAB License Number: 40523909
Operating System: Microsoft Windows 7 Enterprise  Version 6.1 (Build 7601: Service Pack 1)
Java Version: Java 1.8.0_121-b13 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 9.3         (R2017b)
Simulink                                              Version 9.0         (R2017b)
Aerospace Blockset                                    Version 3.20        (R2017b)
Aerospace Toolbox                                     Version 2.20        (R2017b)
Antenna Toolbox                                       Version 3.0         (R2017b)
Audio System Toolbox                                  Version 1.3         (R2017b)
Bioinformatics Toolbox                                Version 4.9         (R2017b)
Communications System Toolbox                         Version 6.5         (R2017b)
Computer Vision System Toolbox                        Version 8.0         (R2017b)
Control System Toolbox                                Version 10.3        (R2017b)
Curve Fitting Toolbox                                 Version 3.5.6       (R2017b)
DSP System Toolbox                                    Version 9.5         (R2017b)
Data Acquisition Toolbox                              Version 3.12        (R2017b)
Database Toolbox                                      Version 8.0         (R2017b)
Datafeed Toolbox                                      Version 5.6         (R2017b)
Econometrics Toolbox                                  Version 4.1         (R2017b)
Embedded Coder                                        Version 6.13        (R2017b)
Filter Design HDL Coder                               Version 3.1.2       (R2017b)
Financial Instruments Toolbox                         Version 2.6         (R2017b)
Financial Toolbox                                     Version 5.10        (R2017b)
Fixed-Point Designer                                  Version 6.0         (R2017b)
Fuzzy Logic Toolbox                                   Version 2.3         (R2017b)
Global Optimization Toolbox                           Version 3.4.3       (R2017b)
HDL Coder                                             Version 3.11        (R2017b)
HDL Verifier                                          Version 5.3         (R2017b)
Image Acquisition Toolbox                             Version 5.3         (R2017b)
Image Processing Toolbox                              Version 10.1        (R2017b)
Instrument Control Toolbox                            Version 3.12        (R2017b)
LTE System Toolbox                                    Version 2.5         (R2017b)
MATLAB Coder                                          Version 3.4         (R2017b)
MATLAB Compiler                                       Version 6.5         (R2017b)
MATLAB Compiler SDK                                   Version 6.4         (R2017b)
MATLAB Report Generator                               Version 5.3         (R2017b)
Mapping Toolbox                                       Version 4.5.1       (R2017b)
Model Predictive Control Toolbox                      Version 6.0         (R2017b)
Model-Based Calibration Toolbox                       Version 5.3         (R2017b)
Neural Network Toolbox                                Version 11.0        (R2017b)
OPC Toolbox                                           Version 4.0.4       (R2017b)
Optimization Toolbox                                  Version 8.0         (R2017b)
Parallel Computing Toolbox                            Version 6.11        (R2017b)
Partial Differential Equation Toolbox                 Version 2.5         (R2017b)
Phased Array System Toolbox                           Version 3.5         (R2017b)
Polyspace Bug Finder                                  Version 2.4         (R2017b)
Polyspace Code Prover                                 Version 9.8         (R2017b)
Powertrain Blockset                                   Version 1.2         (R2017b)
RF Blockset                                           Version 6.1         (R2017b)
RF Toolbox                                            Version 3.3         (R2017b)
Risk Management Toolbox                               Version 1.2         (R2017b)
Robotics System Toolbox                               Version 1.5         (R2017b)
Robust Control Toolbox                                Version 6.4         (R2017b)
Signal Processing Toolbox                             Version 7.5         (R2017b)
SimBiology                                            Version 5.7         (R2017b)
SimEvents                                             Version 5.3         (R2017b)
Simscape                                              Version 4.3         (R2017b)
Simscape Driveline                                    Version 2.13        (R2017b)
Simscape Electronics                                  Version 2.12        (R2017b)
Simscape Fluids                                       Version 2.3         (R2017b)
Simscape Multibody                                    Version 5.1         (R2017b)
Simscape Power Systems                                Version 6.8         (R2017b)
Simulink 3D Animation                                 Version 7.8         (R2017b)
Simulink Check                                        Version 4.0         (R2017b)
Simulink Code Inspector                               Version 3.1         (R2017b)
Simulink Coder                                        Version 8.13        (R2017b)
Simulink Control Design                               Version 5.0         (R2017b)
Simulink Coverage                                     Version 4.0         (R2017b)
Simulink Design Optimization                          Version 3.3         (R2017b)
Simulink Design Verifier                              Version 3.4         (R2017b)
Simulink Desktop Real-Time                            Version 5.5         (R2017b)
Simulink PLC Coder                                    Version 2.4         (R2017b)
Simulink Real-Time                                    Version 6.7         (R2017b)
Simulink Report Generator                             Version 5.3         (R2017b)
Simulink Requirements                                 Version 1.0         (R2017b)
Simulink Test                                         Version 2.3         (R2017b)
Spreadsheet Link                                      Version 3.3.2       (R2017b)
Stateflow                                             Version 9.0         (R2017b)
Statistics and Machine Learning Toolbox               Version 11.2        (R2017b)
Symbolic Math Toolbox                                 Version 8.0         (R2017b)
System Identification Toolbox                         Version 9.7         (R2017b)
Tracking and Sensor Fusion Toolbox                    Version 1.0         (R2017b)
Trading Toolbox                                       Version 3.3         (R2017b)
Vehicle Network Toolbox                               Version 3.4         (R2017b)
Vision HDL Toolbox                                    Version 1.5         (R2017b)
WLAN System Toolbox                                   Version 1.4         (R2017b)
Wavelet Toolbox                                       Version 4.19        (R2017b)
>> 
#### Output of `gpuDevice`
no GPU
#### The command you used
(e.g., `jrc detect-sort myparam.prm`)
jrc manual
#### If manually sorting, what it is you tried to do
splitting
#### Console output

aliddell, 2019-07-24:
Where do you get that impression? By my reading, the split dialog is showing the mean waveform on the current site in the curation GUI, which is the max site unless you've manually switched sites in the time view. The splitting itself, provided you use multisite (which is the default), is done using the traces on all sites in the unit's site neighborhood.

thomasslai, 2019-07-29:
Thanks for the reply!
The default site in the time view was site 1 here. I switched it to site 6 for demonstration purposes, but in the split dialog it still only displays site 1, or at least that's what it says there (see green circles).

[image]

aliddell, 2019-07-29:
Thanks for the update. I see I passed the label for the wrong site, but it does indeed show the mean waveform on the currently-selected site. You can tell by looking at the mean waveform in the split figure, and seeing that it looks much more like the one on site 6 than the one on site 1.

The real issue is why we have site 1 as the primary for this cluster and not site 3. Can you do a histogram of `spikeSites` for spikes in this cluster?

thomasslai, 2019-07-30:
I wasn't sure how to select `spikeSites` for a specific cluster, so here's the histogram of the full `res.spikeSites`. Apparently it only contains a subset of the available sites and it looks like these are those sites with the lowest sitenr. of the group of clusters, e.g. 1 for the first group, 9 for the second group, 17 for third etc (see screenshots).

[image]

[image]

[image]

aliddell, 2019-08-01:
That's pretty strange. Can I see the data/prm file?

thomasslai, 2019-08-07:
Sorry for the delayed response. You can find the .dat and the .prm file under the following link:

https://www.dropbox.com/sh/sm3y3vjn6j4rc1l/AADVYjv0AaA9DxmT0y9knw3ya?dl=0

Thanks!