JaneliaSciComp/JRCLUST #131: issues with curation of non-contiguous recordings

agbondy, 2019-09-20:
I have started doing Neuropixels 1.0 recordings using a checkerboard pattern of recordings that spans multiple banks. My pipeline involves sorting with kilosort2 and curating in JRClust. The sorting and importing works fine, but the case of non-contiguous recordings doesn't seem to have been implemented for curation. I'm attaching a screenshot of what things look like, with the units sorted by centroid. As you can see from the "Unit Positions" and "Probe Map" windows, JRClust knows where the recorded channels are located spatially, and even picks up how the columns being recorded shift at the bank boundary. But this information isn't being used in the Average Waveform and Feature Projection windows. The average waveform y-axis is still sorted according to acquisition channel number, not from bottom to top as would make more sense. As a result, there are two diagonal bands (one for bank 0, one for bank 1) with lots of missing sites in between. (Sorting by center site changes the x-axis order but not the y-axis order so doesn't solve the problem). This could easily be fixed by "reordering" the site numbers in order along the probe, rather than in the order of acquisition. The same problem is visible in the Feature Projection window, where the neighboring sites are not actually contiguous along the probe. Similarly, the "left" and "right" buttons in the figtime window select neighboring acquisitions channels, not neighboring sites on the probe. This makes curation considerably more difficult for non-contiguous recordings.

[image]

aliddell, 2019-09-23:
I'm not sure what you mean by a non-contiguous recording here. Do you mean that you have multiple probes? From your mean waveform window it looks like you're mapping a given site to two different x-y positions, but I could be misreading that. How are you determining your site ordering, and how are you configuring which sites are neighbors?

BTW if you ever feel like committing back upstream, PRs are always welcome. :^)

agbondy, 2019-09-23:
This is a single-shank Neuropixels 1.0 probe.  By non-contiguous, I mean that the recording spans multiple banks, in this case using a checkerboard arrangement (for this recording I used SpikeGLX with an IMRO table provided by Bill Karsh at ). The simplest example of such a "non-contiguous" recording arrangement would be to configure the probe to record as follows: channel 1 bank 0, channel 2 bank 1, channel 3 bank 0, channel 4 bank 1, etc. This takes advantage of the Neuropixel probe's ability to have a single recording channel address up to three sites to cover a larger distance, albeit with reduced site density. This requires "non-contiguity" in the sense that neighboring recording channels are addressing different banks and therefore non-neighboring sites on the probe. All the information about the actual recording sites being addressed by each recording channel in the data stream is passed to JRClust using the siteLoc parameter in the .prm file, and physical neighbors are extracted correctly given this information. The only problem is that a subset of plots being generated (the mean waveform plot, the time plot and the feature projection plot) are not using the siteLoc information to index recording sites, instead just plotting the recording channels in order, introducing some undesirable plotting behavior. The presence of two diagonal bands in the mean waveform plot is an example of this. What is happening (I think) is that the cluster centroids (on the x-axis) are ordered according to the physical arrangement of cluster centroids on the probe, while the recording sites (y-axis) are indexed according to the order of the channels in the data stream, which is not in physical order on the probe. I have a pretty good sense of what needs to be changed to fix this, but probably won't get around to it for several weeks so I thought I would share the issue. Happy to send over more PRs! I think JRClust is great and it's cool that you are opening to folks contributing.

closed 2020-03-18 by agbondy