JaneliaSciComp/JRCLUST #135: Clusters are not ordered by center site and cannot reorder

LiuDaveLiu, 2019-11-19:
If you have a usage or support question, come join our [Gitter](https://gitter.im/JRCLUST/community) and ask your question there.
If you found a bug, please provide **all** of the following:

#### Output of `jrc version`
JRCLUST v4.0.1 "Edward" (f951f8e)

#### Output of `gpuDevice` (if you're seeing a GPU error)

#### The command you used
manual sorting
(e.g., `jrc detect-sort myparam.prm`)

#### If manually sorting, what it is you tried to do
Recorder cluster by center site

#### Console output
Index exceeds the number of array elements (400).

Error in jrclust.interfaces.Clustering/orderClusters>@(vals,indices)vals(indices) (line 26)
    hFunSubset = @(vals, indices) vals(indices);

Error in jrclust.interfaces.Clustering/orderClusters (line 35)
            metadata.(fn) = hFunSubset(metadata.(fn), argsort);

Error in jrclust.curate.CurateController/reorderClusters (line 7)
    argsort = obj.hClust.orderClusters(by);

Error in jrclust.curate.CurateController/addMenu>@(hO,hE)obj.reorderClusters('clusterSites') (line 32)
    uimenu(obj.hMenus('EditMenu'), 'Label', 'Reorder clusters by center site', 'Callback', @(hO, hE) obj.reorderClusters('clusterSites'),
    'Separator', 'on');
 
Error while evaluating Menu Callback.

aliddell, 2019-11-19:
I think this is fixed in the latest commit. Can you pull and try again?

LiuDaveLiu, 2019-11-19:
New error,

Error using reshape
To RESHAPE the number of elements must not change.

Error in JRC/loadFiles>readBin (line 168)
        binData = reshape(binData, binShape);

Error in JRC/loadFiles (line 25)
            spikesRaw = readBin(obj.hCfg.rawFile, res_.rawShape, '*int16');

Error in JRC/curate (line 4)
        obj.loadFiles();

Error in JRC/run (line 27)
        obj.curate();

Error in jrc (line 20)
        hJRC.run();

aliddell, 2019-11-19:
Delete your .mat file and try again, please.

LiuDaveLiu, 2019-11-19:
Sorry, which .mat file? Not the _res.mat, right? because it won't run without that.

LiuDaveLiu, 2019-11-20:
I resorted the data on another computer, and got this error after processing the last file. I suspect it's something with batch sorting multiple files.

Error using reshape
To RESHAPE the number of elements must not change.

Error in jrclust.detect.DetectController/detect (line 105)
    res.spikesRaw = reshape(fread(fid, inf, '*int16'), res.rawShape);

Error in JRC/detect (line 16)
    dRes = obj.hDetect.detect();

Error in JRC/run (line 13)
        obj.detect();

Error in jrc (line 20)
        hJRC.run();

aliddell, 2019-11-20:
Fixed in person.

closed 2019-11-20 by aliddell