JaneliaSciComp/JRCLUST #136: GPU filtering error

LiuDaveLiu, 2019-11-19:
If you have a usage or support question, come join our [Gitter](https://gitter.im/JRCLUST/community) and ask your question there.
If you found a bug, please provide **all** of the following:

#### Output of `jrc version`
JRCLUST v4.0.1 "Edward" (3d2e75c)
#### Output of `gpuDevice` (if you're seeing a GPU error)
CUDADevice with properties:

                      Name: 'GeForce GTX 1080 Ti'
                     Index: 1
         ComputeCapability: '6.1'
            SupportsDouble: 1
             DriverVersion: 10.1000
            ToolkitVersion: 10
        MaxThreadsPerBlock: 1024
          MaxShmemPerBlock: 49152
        MaxThreadBlockSize: [1024 1024 64]
               MaxGridSize: [2.1475e+09 65535 65535]
                 SIMDWidth: 32
               TotalMemory: 1.1811e+10
           AvailableMemory: 9.6727e+09
       MultiprocessorCount: 28
              ClockRateKHz: 1670500
               ComputeMode: 'Default'
      GPUOverlapsTransfers: 1
    KernelExecutionTimeout: 1
          CanMapHostMemory: 1
           DeviceSupported: 1
            DeviceSelected: 1
#### The command you used
jrc detect-sort 
(e.g., `jrc detect-sort myparam.prm`)

#### If manually sorting, what it is you tried to do

#### Console output
2019-11-19 15:33:08 Applying bandpass filter to 287868 samples...
2019-11-19 15:33:09 GPU filtering failed: BLAS loading error:
mkl_custom.dll: The specified module could not be found.

 (retrying on CPU)...
2019-11-19 15:33:09 GPU filtering failed: BLAS loading error:
mkl_custom.dll: The specified module could not be found.

 (retrying in CPU)...
2019-11-19 15:33:10 Applying bandpass filter to 287868 samples...
2019-11-19 15:33:10 GPU filtering failed: BLAS loading error:
mkl_custom.dll: The specified module could not be found.

 (retrying on CPU)...
Error using  \ 
BLAS loading error:
mkl_custom.dll: The specified module could not be found.

Error in zp2ss (line 108)
    a1 = t\[-den(2) -den(3); 1 0]*t;

Error in ellip (line 102)
[a,b,c,d] = zp2ss(z,p,k);

Error in jrclust.filters.bandpassFilter>makeFilt_ (line 83)
                [vrFiltB, vrFiltA]=ellip(filtOpts.filtOrder,0.1,40, freqLimBP, vcType);

Error in jrclust.filters.bandpassFilter>filtfiltChain (line 23)
       [cvrB{end+1}, cvrA{end+1}] = makeFilt_(filtOpts.freqLimBP, 'bandpass', filtOpts);

Error in jrclust.filters.bandpassFilter (line 9)
        samplesOut = filtfiltChain(single(samplesIn), hCfg);

Error in jrclust.filters.filtCAR (line 32)
        samplesOut = jrclust.filters.bandpassFilter(samplesOut, hCfg);

Error in jrclust.detect.DetectController/filterSamples (line 16)
        [samplesOut, channelMeans] = jrclust.filters.filtCAR(samplesIn, [], [], 0, obj.hCfg);

Error in jrclust.detect.DetectController/detectOneRecording (line 160)
            [iSamplesFilt, keepMe] = obj.filterSamples(iSamplesRaw, samplesPre, samplesPost);

Error in jrclust.detect.DetectController/detect (line 66)
        recData = obj.detectOneRecording(hRec, [rawFid, filtFid], impTimes, impSites, siteThresh);

Error in JRC/detect (line 16)
    dRes = obj.hDetect.detect();

Error in JRC/run (line 13)
        obj.detect();

Error in jrc (line 20)
        hJRC.run();

aliddell, 2019-11-19:
Is this on that spike sorting machine in Karel's pod?

LiuDaveLiu, 2019-11-19:
Yes

aliddell, 2019-11-19:
Try adding C:\Users\JRCLUST\lib to your path.

LiuDaveLiu, 2019-11-19:
I still have the same error.

aliddell, 2019-11-20:
Either way, this is specific to your setup and not a JRCLUST issue. I'm happy to help (I'm around now actually), but I'm going to close this.

closed 2019-11-20 by aliddell