JaneliaSciComp/JRCLUST #137: Error in autosorting

LiuDaveLiu, 2019-11-29:
If you have a usage or support question, come join our [Gitter](https://gitter.im/JRCLUST/community) and ask your question there.
If you found a bug, please provide **all** of the following:

#### Output of `jrc version`
JRCLUST v4.0.1 "Edward" (3d2e75c)
#### Output of `gpuDevice` (if you're seeing a GPU error)

#### The command you used
jrc sort myparam.prm
(e.g., `jrc detect-sort myparam.prm`)

#### If manually sorting, what it is you tried to do

#### Console output
Error using horzcat
Dimensions of arrays being concatenated are not consistent.

Error in jrclust.features.getSiteFeatures (line 53)
        siteFeatures = [sf1, sf2; single(spikeData.spikeTimes([spikes1; spikes2]))']; % TW

Error in jrclust.sort.computeRho (line 46)
        [siteFeatures, ~, n1, n2, spikeOrder] = jrclust.features.getSiteFeatures(dRes.spikeFeatures, iSite, spikeData, hCfg);

Error in jrclust.sort.SortController/sort (line 44)
            res = jrclust.sort.computeRho(dRes, res, obj.hCfg);

Error in JRC/sort (line 38)
    sRes = obj.hSort.sort(obj.res);

Error in JRC/run (line 20)
        obj.sort();

Error in jrc (line 20)
        hJRC.run();

LiuDaveLiu, 2019-12-03:
% function [mrFet12_, viSpk12_, n1_, n2_, viiSpk12_ord_] = fet12_site_(trFet_spk, S0, P, iSite, vlRedo_spk)
function [siteFeatures, spikes, n1, n2, spikeOrder] = getSiteFeatures(spikeFeatures, site, spikeData, hCfg)
    %GETSITEFEATURES Get features occurring on primary and secondary (optionally tertiary) sites
    [siteFeatures, spikes, n1, n2, spikeOrder] = deal([]);

    nPeaksFeatures = hCfg.nPeaksFeatures; % or maybe size(spikeFeatures, 2) ? 
    timeFeatureFactor = hCfg.getOr('timeFeatureFactor', 0); % TW

    if isfield(spikeData, 'spikes1') && ~isempty(spikeData.spikes1)
        spikes1 = int32(spikeData.spikes1);
    else
        return
    end

    if isfield(spikeData, 'spikes2') && ~isempty(spikeData.spikes2)
        spikes2 = int32(spikeData.spikes2);
    else
        spikes2 = [];
        nPeaksFeatures = 1;
    end

    if isfield(spikeData, 'spikes3') && ~isempty(spikeData.spikes3)
        spikes3 = int32(spikeData.spikes3);
    else
        spikes3 = [];
    end

    if isfield(spikeData, 'vlRedo_spk') && ~isempty(spikeData.vlRedo_spk)
        spikes1 = spikes1(spikeData.vlRedo_spk(spikes1));
        spikes2 = spikes2(spikeData.vlRedo_spk(spikes2));
        spikes3 = spikes3(spikeData.vlRedo_spk(spikes3));
    end

    n1 = numel(spikes1);
    n2 = numel(spikes2);

    % get features for each site
    if nPeaksFeatures == 1
        sf1 = squeeze(spikeFeatures(:, 1, spikes1));
        if iscolumn(sf1), sf1 = sf1'; end

        siteFeatures = [sf1; single(spikeData.spikeTimes(spikes1))']; % TW
        siteFeatures(end, :) = timeFeatureFactor*std(siteFeatures(1, :)).*siteFeatures(end, :)./std(siteFeatures(end, :)); % TW

        spikes = spikes1;
    elseif nPeaksFeatures == 2
        sf1 = squeeze(spikeFeatures(:, 1, spikes1));

        sf2 = squeeze(spikeFeatures(:, 2, spikes2));
        if iscolumn(sf1) && iscolumn(sf2)
            sf1 = sf1';
            sf2 = sf2';
        end

        siteFeatures = [sf1, sf2; single(spikeData.spikeTimes([spikes1; spikes2]))']; % TW
        siteFeatures(end, :) = timeFeatureFactor*std(siteFeatures(1, :)).*siteFeatures(end, :)./std(siteFeatures(end, :)); % TW

        spikes = [spikes1; spikes2];
    else % nPeaksFeatures == 3
        sf1 = squeeze(spikeFeatures(:, 1, spikes1));
        if iscolumn(sf1), sf1 = sf1'; end

        sf2 = squeeze(spikeFeatures(:, 2, spikes2));
        if iscolumn(sf2), sf2 = sf2'; end
        
        sf3 = squeeze(spikeFeatures(:, 3, spikes3));
        if iscolumn(sf3), sf3 = sf3'; end
        
        siteFeatures = [sf1, sf2, sf3; single(spikeData.spikeTimes([spikes1; spikes2; spikes3]))'];
        siteFeatures(end, :) = timeFeatureFactor*std(siteFeatures(1, :)).*siteFeatures(end, :)./std(siteFeatures(end, :)); % TW

        spikes = [spikes1; spikes2; spikes3];
        n2 = n2 + numel(spikes3);
    end

    % weight features by distance from site, greater for nearer sites
    try
        nSites = hCfg.nSitesEvt;

        if hCfg.weightFeatures && nSites >= hCfg.minSitesWeightFeatures
            nFeaturesPerSite = size(siteFeatures, 1) / nSites;

            weights = distWeight(site, nSites, hCfg);
            weights = repmat(weights(:), [nFeaturesPerSite, 1]);

            siteFeatures = bsxfun(@times, siteFeatures, weights(:));
        end
    catch ME
        warning('error in distWeight: ''%s'', using unweighted features', ME.message);
    end

    spikeOrder = jrclust.utils.rankorder(spikes, 'ascend');
end

%% LOCAL FUNCTIONS
function weights = distWeight(site, nSites, hCfg)
    %DISTWEIGHT Compute weights for neighboring sites, larger for nearer
    siteLoc = hCfg.siteLoc(hCfg.siteNeighbors(1:nSites, site), :);
    siteDists = pdist2(siteLoc(1, :), siteLoc);

    weights = 2.^(-siteDists/hCfg.evtDetectRad);
    weights = weights(:);
end

LiuDaveLiu, 2019-12-03:
the function is in C:\Github\JRCLUST\+jrclust\+features

LiuDaveLiu, 2019-12-12:
I got a very similar error, can you please take a look?

Error using vertcat
Dimensions of arrays being concatenated are not consistent.

Error in jrclust.features.getSiteFeatures (line 42)
        siteFeatures = [sf1; single(spikeData.spikeTimes(spikes1))']; % TW

Error in jrclust.sort.computeRho (line 46)
        [siteFeatures, ~, n1, n2, spikeOrder] = jrclust.features.getSiteFeatures(dRes.spikeFeatures, iSite, spikeData, hCfg);

Error in jrclust.sort.SortController/sort (line 44)
            res = jrclust.sort.computeRho(dRes, res, obj.hCfg);

Error in JRC/sort (line 38)
    sRes = obj.hSort.sort(obj.res);

Error in JRC/run (line 20)
        obj.sort();

Error in jrc (line 20)
        hJRC.run();

DJ-Hayden, 2020-07-28:
I figured this out (if you're still working on it).
```
if iscolumn(sf1), sf1 = sf1'; end
```
should be
```
if isrow(sf1), sf1 = sf1'; end
```

Same for the sf2 line below that.