JaneliaSciComp/JRCLUST #139: Issue importing Intan Data

DJ-Hayden, 2020-01-27:
#### Output of `jrc version`
JRCLUST v4.0.1 "Edward" (3d2e75c)

#### Output of `gpuDevice` (if you're seeing a GPU error)
Not a GPU error, but it's a GeForce GTX 1070

#### The command you used
`jrc bootstrap`
Then select No I don't have a Meta file, then select the Intan info.rhd file from something that was recorded in 1 channel = 1 file mode.

#### Error that pops up
```
Undefined variable "obj" or class "obj.bootstrapIntan".

Error in JRC/bootstrap>getMetafile (line 194)
                obj.bootstrapIntan(binfile);

Error in JRC/bootstrap (line 41)
        [metafile, binfile, workingdir] = getMetafile(workingdir);

Error in JRC/processArgs (line 127)
            obj.bootstrap(obj.args{:});

Error in JRC (line 54)
                    obj.processArgs();

Error in jrc (line 16)
    hJRC = JRC(varargin{:});
```

aliddell, 2020-03-06:
Well, this is embarrassing. I'm waiting on some changes to a pull request that will change `bootstrap.m`, so I don't want to just push a fix that breaks the PR.

For right now you should be able to do `jrc bootstrap /path/to/file.rhd`, and it should Just Work (TM). Let me know if that's not the case.

DJ-Hayden, 2020-03-07:
It does work, but then I think it's assuming I recorded everything in matrix format (as opposed to the one file per channel format). Regardless, I switched to a simpler version wherein I just create a matrix of int16s that correspond to the probe file. Then I feed that .dat file into JRClust using the instructions.

DJ-Hayden, 2020-03-07:
Though as an aside, I store the data as int16's. To convert to microvolts, you multiply that value by 0.195 (per Intan's documentation for my specific recording set up). Assuming that is true, is 0.195 the uV/bit value within the JRClust prm file?

aliddell, 2020-03-09:
When I was working on the Intan format support, I remember there being a "traditional" format and a newer one. Is the one file/channel format the newer one?

> Assuming that is true, is 0.195 the uV/bit value within the JRClust prm file?

It depends on your filter parameter. If your filter type is 'ndiff' (which is the default), then it needs to be 0.39, or twice 0.195. If your filter type is 'sgdiff', then it needs to be 0.39 * `(sum(1:nDiffOrder).^2)`. Otherwise you can use 0.195.

DJ-Hayden, 2020-03-10:
I believe the one file/channel format is the newer one. Would also explain why some of their documentation about it has typos/is wrong. Filter type is the default, so I'll just update that in my next round. Thanks!

closed 2020-03-10 by DJ-Hayden