JaneliaSciComp/JRCLUST #149: spikeTimes missmatch + Incorrect opening and plot of extracted spike traces

franciscamachado, 2020-05-08:
Hello,

After clustering spikes and curating the clusters of a dat.file I saved the results to a .csv file. However when i plot the spikes in the original signal according to the spikeTimes presented in the .csv file, these do not match with the original signal's spikes. The spikes' negative dip is marked incorrectly (indicating a   timing missmatch). 

Secondly, i tried opening the configfile_raw.jrc binary file with the raw spike traces. Yet, the result was an array with nlines x 1 column, instead of being a nsamples×nsites×nspikes  array. Do i need to specify more information about the file structure to open it correctly? I opened the file like this:
'''
fid = fopen('configfile_raw.jrc', 'r');
waveforms= fread(fid, inf, '*int16');
fclose(fid);
'''

Lastly, I also have an issue in plotting the exported spike traces (GUI output) of a specific cluster for the site with the best waveforms (as seen in the GUI), considering that the results do not match with the ones presented in the GUI's waveform view and are very polluted with other waveforms that do not belong to the cluster in question.

How can i solve these problems?

Thank you.

aliddell, 2020-05-12:
A couple of questions:

How are you doing the export to CSV? Are you using the JRCLUST exporter? How are you then plotting traces from the raw data? How is the negative dip marked incorrectly? Do you see this same issue in the GUI?

The configfile_raw.jrc binary has no intrinsic shape, i.e., if you open it with just `fopen` and read with `fread`, you'll get a 1D array of integers. You need to use the [reshape](https://www.mathworks.com/help/matlab/ref/reshape.html) command to put the traces in the right order. The [proper shape](https://jrclust.readthedocs.io/en/latest/usage/io.html#io-raw-traces) is a value in the .mat file, `rawShape`.

> Lastly, I also have an issue in plotting the exported spike traces (GUI output) of a specific cluster for the site with the best waveforms (as seen in the GUI), considering that the results do not [...]

Maybe it would be best if you could share the code you use to do this, or maybe working together in real time would be an even better way. [Let me know!](mailto:alan@vidriotech.com)

franciscamachado, 2020-05-15:
Hello,

Thank you very much. I was able to open the files with the reshape command.

As for the timing missmatch, when i take the spiketimes (either from the .csv file exported from the GUI or from the continuous_res "spikeTimes" variable) and plot these points in the original signal they are never at the negative peak (as seen in the image below).
[image]