JaneliaSciComp/JRCLUST #151: Bug - Error message after GUI re-opening and after splitting clusters

franciscamachado, 2020-05-22:
Hello,

After saving the GUI results for the first time and after splitting clusters, I am unable to re-open  the results on the GUI ( " jrc manual configfile.prm "), for which I receive an error message and all the clusters' waves appear in gray, while being unable to work with the GUI (as seen in the first image)
[image]

Furthermore, after saving the GUI results for the first time without conducting any cluster splitting, when i open the GUI ( " jrc manual configfile.prm ") once again I am unable to re-organize the sites by center sites, for which i receive an error message (as seen in the second figure).
[image]

However, after several attempts i discovered that if I do not save the results after performing the clustering (" jrc full configfile.prm ") and curation with the GUI (" Exit -> Save clustering to... -> NO " ) for the first time and if I re-open the results (" jrc manual configfile.prm "), the sites and clusters are organized and I can make splits and then save the results, while being able to re-open the GUI without any error messages. So its a question of not saving the results for the first time in the GUI.

This happened in multiple computers and in all recording sessions.

gombpeti, 2020-08-06:
Hi,
I have the same problem. But I would like to be more accurate in the symptoms:
After manual clustering -> I go to "Exit" -> pop up automatically: "do you want to save..." -> click to OK -> next time the clusters organized well for me and I can split.
BUT -> during the manual clustering I go manually to File -> save and then the next time when I open my data: THE PROBLEM START. The clustering completely wrong and then I ask reclustering ->  I got msg: "operation failed" the Autocorr is also looking bad. 
I hope it helps to discover the problem.
P

aliddell, 2020-11-05:
@franciscamachado @gombpeti do you still see this in the latest commit?

LJ-lab, 2020-12-16:
The problem still persists with the new commit. The sites still show up in scrambled positions if one saves the first time it runs the full clustering sequence (jrc full) and opens it again (with jrc manual).

And now there is an additional error we had never encountered before. If we now run the full sequence, then do some merging or splitting, save, close, and then re-open with jrc manual we are getting the following error: "Data found to be in an inconsistent state".